cleanflight/cleanflight #1559: PWM is initialised before the I2C bus

martinbudden, 2015-12-10:
In particular, in the `init()` function in `main.c`, `pwmInit` is called before `i2cInit`. This means that a sensor that could be either use the `timerHardware` or the I2C bus (eg sonar) cannot be detected and initialised properly.

Currently if `FEATURE_SONAR` is enabled then `timerHardware` has to be reserved, even if it turns out the user has attached an I2C based sonar device.

Is it possible for i2cInit to be called earlier?

hydra, 2015-12-13:
pwmInit() needs to be called asap for ESC compatibility reasons.  See #357

hydra, 2015-12-13:
At the moment there is nothing to do for this ticket because there is no reason to change things for the existing code.  If you have I2C sensor code to add to the codebase then you can address this as part of that feature.  Just be sure not to delay pwm initialisation when you do so.   I'll close this ticket and wait for a PR that needs and contains the appropriate changes.

closed 2015-12-13 by hydra

martinbudden, 2015-12-13:
OK, that makes sense.

I'm working on some IC2 sonar code where this would be a "nice to have" rather than being a necessity. So I'm not going to propose any changes as part of that.

In the meantime I've created a pull request with a comment so that this is documented in the code:

https://github.com/cleanflight/cleanflight/pull/1567