cleanflight/cleanflight #1560: Why not allowing LED_STRIP together with SOFTSERIAL on NAZE?

krzysztofmatula, 2015-12-11:
SOFTSERIAL feature can be useful (e.g. for telemetry). But I guess 2 SOFTSERIAL ports are used less often. Why not allowing to enable a single SOFTSERIAL, leaving pins for other cool features like LED_STRIP or current meter?

dkisselev, 2015-12-11:
No, it's not currently possible on F1 boards.

Both of the features use a hardware timer module on the CPU, which there are a limited number of. Two softserials can be powered by a single timer, but with the way everything is currently  implemented  we can't have a softserial and LED strip running at the same time.

With the rise of F3 boards, I'm not sure that the demand will be there to rewrite that much of the logic to add support, either.

hydra, 2015-12-12:
@dkisselev is correct.  Not currently possible.  There are other PR's to improve softserial and it's something we could look at in the future.  For now the answer is use an F3 board which have more hardware serial ports as standard.

closed 2015-12-12 by hydra