cleanflight/cleanflight #1561: Mixer for A-tail4 looks wrong

martinbudden, 2015-12-12:
The mixer for the A-tail4 looks incorrect. All the values are unitary. Because of the angled motors on the A-tail I would expect some of the values to be fractional. Indeed I would expect the Atail4 matrix to be a permutation of the V-tail4 matrix. (My assumption here is that A-tail4 means the configuration that is like an inverted V-tail, if it means something else then I am barking up the wrong tree.)

(I guess that the Atail4 matrix was put in as an initial placeholder and hasn't yet been adjusted.)

In `mixer.c`:

    static const motorMixer_t mixerAtail4[] = {
    { 1.0f,  0.0f,  1.0f,  1.0f },          // REAR_R
    { 1.0f, -1.0f, -1.0f,  0.0f },          // FRONT_R
    { 1.0f,  0.0f,  1.0f, -1.0f },          // REAR_L
    { 1.0f,  1.0f, -1.0f, -0.0f },          // FRONT_L
    };
for comparison

    static const motorMixer_t mixerVtail4[] = {
    { 1.0f,  -0.58f,  0.58f, 1.0f },        // REAR_R
    { 1.0f,  -0.46f, -0.39f, -0.5f },       // FRONT_R
    { 1.0f,  0.58f,  0.58f, -1.0f },        // REAR_L
    { 1.0f,  0.46f, -0.39f, 0.5f },         // FRONT_L
    };

hydra, 2015-12-12:
Please can you submit a PR with set of suitable values?

martinbudden, 2015-12-13:
OK, I'll look into this with a view to submitting a PR.

I don't have an A-tail, so I can't test this. I am very reluctant to submit untested code, so I'd like to make the following provisos:

1. You thoroughly check submission
2. You put a warning in the release notes that the new A-tail mixer is experimental and ask any users to report any problems

My reluctance to submit untested code is somewhat mitigated by the fact that it looks like the current mixer is broken and that it is probably not widely used.

Before I make any changes I need to confirm the motor order. The cleanflight configurator file:

https://github.com/cleanflight/cleanflight-configurator/blob/master/resources/motor_order/atail_quad.svg

has a non-standard motor order. It is the only quad configuration with that motor order, so I suspect it is in error. Can you check this?

martinbudden, 2015-12-13:
OK, here's my mixer for the A-tail. Note that the motors rotate in the opposite sense to the V-tail (and so in the same sense as the QuadX and QuadP). Compared with V-tail are we have:

1) throttle unchanged
2) orientation of motors on tail (pointing outwards or pointing inwards) does not actually effect roll, so roll unchanged
3) pitch unchanged
4) yaw sense reversed

I've got the code ready. Can you double check this and also confirm the motor order (as per my previous post) and then I'll submit my pull request.

    /* Atail4
     4CW----2CCW
         ||
         ||
         /\
    3CCW/  \1CW
    */
    static const motorMixer_t mixerAtail4[] = {
        throttle, roll,  pitch,   yaw
        { 1.0f, -0.58f,  0.58f, -1.0f },        // REAR_R
        { 1.0f, -0.46f, -0.39f,  0.5f },        // FRONT_R
        { 1.0f,  0.58f,  0.58f,  1.0f },        // REAR_L
        { 1.0f,  0.46f, -0.39f, -0.5f },        // FRONT_L
    };

For comparision, here's the V-tail
    
    /* Vtail4
    4CCW-----2CW
         ||
         ||
    3CW\ || /1CCW
        \||/
    */
    static const motorMixer_t mixerVtail4[] = {
        throttle, roll,  pitch,   yaw
        { 1.0f, -0.58f,  0.58f,  1.0f },        // REAR_R
        { 1.0f, -0.46f, -0.39f, -0.5f },        // FRONT_R
        { 1.0f,  0.58f,  0.58f, -1.0f },        // REAR_L
        { 1.0f,  0.46f, -0.39f,  0.5f },        // FRONT_L
    };

hydra, 2016-01-12:
PR please!

hydra, 2016-05-07:
#1674 merged to fix.

closed 2016-05-07 by hydra