cleanflight/cleanflight #1562: No mixer for spider quadcopter frame type

martinbudden, 2015-12-12:
The spider configuration is an H configuration where the front arms are rotated slightly backwards to move the propellers out of view of a front mounted camera, e.g.:

    4CW-----2CCW
        ||
        /\
       /  \
    3CCW   1CW

Examples of this frame type include the Team Black Sheep Discovery (a 450 class frame) and the Diatone ET180 (a 180 class frame).

Spider configurations vary in the amount the front arms have been rotated, but a generic spider mixer would still be a better fix than the Quad-X mixer.

Here's one I made earlier:

    static const motorMixer_t mixerQuadSpider[] = {
        { 1.0f, -1.0741f,  1.0943f, -1.0513f },          // REAR_R
        { 1.0f, -0.8056f, -1.0943f,  0.9535f },          // FRONT_R
        { 1.0f,  1.0741f,  1.0943f,  1.0513f },          // REAR_L
        { 1.0f,  0.8056f, -1.0943f, -0.9535f },          // FRONT_L
    };

joshuabardwell, 2015-12-12:
I'm curious whether it can be demonstrated that a custom mix for a spider quad is actually necessary. I believe it was QuadMcFly on RCGroups (not 100% sure my memory is correct about who it was) who demonstrated mathematically that a motor's distance from CG did not need to be corrected for with a cmix, because the different arc length through which the motor needed to travel was exactly compensated for by the different torque that resulted from the motor's distance from CG. In other words, a motor that is farther from the CG must travel a longer distance to effect a given number of degrees of rotation, but it has a longer lever arm and so exerts more torque than a motor further in. Therefore, it moves further for a given amount of force applied.

A spider hex would be a perfect vehicle for testing this premise. The fact that the rear motors are closer to the roll axis than the front motors would mean that the copter would tend to pitch slightly on roll input, as the rear motors either lagged or led the front motors. If a properly tuned spider quad does not exhibit this behavior, then it supports the premise.

martinbudden, 2015-12-12:
OK, you've got me thinking.

Two thoughts here:
a) A "true" spider quad, that it one which takes a quadX and just rotates the forward arms back a bit, and rotates the back arms in a bit, does not actually change the distance of any of the motors from the CG.

b) But something is going on. Consider this: take an X quad and rotate the front right arm counter-clockwise until it points directly forward. By your above argument the mixer does not need to change. Then take the front left arm and rotate it counter-clockwise until it points directly left. Still no need to to change the mixer. I think you can see where I am going. Then rotate the back left arm and then the back right arm. We've transformed our quadX into a quadPlus, and this certainly requires a change in the mixer.

So there is a flaw in someone's argument, either mine or QuadMcFly's.

I think the issue is that it's not the change of distance of the motors from the CG that causes a need for a new mixer, but the change in angles of the lines between the motors and the CG. The overall effect of the motor is unchanged, but the roll and pitch components vary.

Anyway, that's my current thinking, but I'm prepared to be convinced otherwise.

joshuabardwell, 2015-12-12:
If you use board_align=45 on an X frame, it would fly exactly as if it was a plus frame, no custom mixer required.

joshuabardwell, 2015-12-12:
I think the relevant factor is the distance of the motor from the axis of rotation, not from the CG. In a rectangle quad, the motors are equidistant from the axis of rotation and so no custom mix is required. In a spider quad, the rear motors are closer to the roll axis, and a custom mix may be required. This is what I'm curious about.

martinbudden, 2015-12-13:
Yes, that is true. But isn't using yaw board align equivalent to using a mixer on the input stage (as opposed to on the output stage)?

Let's do another thought experiment. Let's take our QuadX and vertically displace the back right arm vertically a bit. This should have negligable effect. Then rotate the both the back arms so that we have a Y4 configuration. This definitely requires a new mixer and cannot be compensated by changing the aligment of the frame.

The more I think about this the more I think that the argument you quoted is not correct. However I must admit I have been doing thought experiments rather than sitting down to do the maths.

joshuabardwell, 2015-12-14:
I'm not convinced that a Y4 needs a custom mix, although I haven't personally tested this, and I respect the experience of anybody who has. What will a Y4 do with a standard QuadX mix? If you roll the copter, one of the rear motors will spin down, and the other rear motor will spin up by the same amount. Since both motors have zero moment arm relative to the roll axis, they will exert no roll force at all. They will also exert no pitch force (in a perfect world) because they will exactly cancel each other out.

This is not to say that you wouldn't benefit from mixing zero roll into the rear motors, just to avoid the real-world destabilization that may come from the motors not exactly spinning up and down together, and also to prevent the motors from reaching the limits of their command authority if/when they min/max out in an aggressive roll. But the underlying premise, that the length of moment arm exactly compensates for the length of arc distance that the motor must travel, is not disproven by the example of the Y4 copter.

The question that still stands, IMO, is: if the Y4 is an extreme example, where mixing 0 roll into the rear motors makes sense, and the QuadX is the opposite extreme, then shouldn't there be a range of motor positions with optimal mixes from 100% to 0%? And if so, what is the relevant parameter that determines the optimal mix. I don't think it's "distance from CG".

martinbudden, 2015-12-14:
I think I understand the issue now.

However I am a little uncomfortable having this discussion in an "issue" thread since it is rather tangental to the original issue. I'm also new to Cleanflight. Is there somewhere more appropriate we can continue this discussion?

martinbudden, 2015-12-22:
I'm withdrawing this I've looked at the measurements of several spider class frames and the configurations do vary, so it is unclear whether a generic spider mixer would be useful.

closed 2015-12-22 by martinbudden