cleanflight/cleanflight #1570: MSP telemetry corrupted (UART overflows?)

krzysztofmatula, 2015-12-13:
I'm experimenting with the "MSP telemetry" feature. I've configured one of HW UARTs for such telemetry and then I wanted to capture the serial output on a PC. I experimented with different baud rates and whatever I did I was getting corrupted output. First I thought is was an issue with my serial-to-usb converter, but finally I've realized that I'm getting the telemetry stream at ~115kbps rate!!! Evidently, the stream was utilizing full UART bandwidth and, probably even worse, it was output even faster resulting in TX buffer overflows which were turning into the data corruption...

The MSP telemetry shall be time-synchronized, resulting in messages being sent at some fixed interval. Additionally, It would be nice to check for TX buffer overflows and suspend the loop until pending messages are sent out.

Additionally, I see that the MSP_BAT messages are not sent in the telemetry. It would be nice to include them as well.

I'm using 1.11 release on NAZE32.

krzysztofmatula, 2015-12-14:
At first I wanted to propose a fix for this issue, but I've just found discussion about removing this feature completely. Using bi-directional MSP is fine for me (maybe even better), if only it worked over soft-serial (issue #1558)... Should I bother fixing the MSP telemetry?

joshuabardwell, 2015-12-14:
MSP does work over soft-serial, most of the time. Occasionally, the FC board hangs (I have an issue open for it).

krzysztofmatula, 2015-12-14:
For the time being (until MSP telemetry is thrown out) I've proposed some changes that make it actually work: https://github.com/cleanflight/cleanflight/pull/1578

krzysztofmatula, 2015-12-15:
MSP telemetry feature is now removed.

closed 2015-12-15 by krzysztofmatula