cleanflight/cleanflight #1572: transmitter range problem

MauroMombelli, 2015-12-14:
hi,
testing the 1.10 i have a problem with some reciver/transmitter pair:
some of them has a range a little above 2100 for a channel, and while i had no problem before, now when a bit above 2100 it automatically reset the channel to 1500.

As this is the default setting of the tx/rx settings, maybe this limit should be incremented to 2500?

joshuabardwell, 2015-12-14:
This is not a bug in cleanflight. Your channels must be within the parameters rx_min_usec and rx_max_usec. Adjust your endpoints in your transmitter to the standard value of 1000/2000 which CleanFlight expects. Or, if you prefer, adjust rx_min_usec and rx_max_usec and use the rxrange command to cause CleanFlight to be synced with your transmitter's existing endpoints.

SteveAmor, 2016-01-08:
@MauroMombelli can you please close this suport request.

closed 2016-01-10 by MauroMombelli