cleanflight/cleanflight #1582: What about supporting LLIGHTS feature?

krzysztofmatula, 2015-12-15:
The BOXLLIGHTS exists in the Cleanflight code, however doesn't seem to be used.
What about using this to switch one of the spare GPIO pins, like PA15 or PB5 on NAZE board?

The LLIGHTS stands of "landing lights", but can be used as a general purpose switch.

handsomejackuk, 2015-12-16:
agree was used in multiwii.... and i did like to use additional lights output, not just for led strips...

krzysztofmatula, 2015-12-16:
Yes, specially when led strip is not available because of using SOFTSERIAL.
I'm just finishing the implementation, stay tuned :)
It will use PA15, positive logic.
Connecting 12V LED will require a serial resistor and NPN transistor (open collector circuit).

JohnOCFII, 2015-12-16:
Sounds interesting.  I too, am out of ports, and would like to be able to use the same LED strings on quads where I can use the Cleanflight (Pin 5) control, as well as via some other method that would be available when I'm using both SoftSerial ports, even if the lights can't be programmatically controlled.

krzysztofmatula, 2015-12-16:
I've implemented it here: https://github.com/krzysztofmatula/cleanflight/commit/a756050a1133a56840bcaeb6e86093598294e86a
Comments welcome.

closed 2017-02-25 by hydra