cleanflight/cleanflight #1586: 1wire support for sparky.

Jahnkeanater, 2015-12-16:
1wire does not work on the sparky board. blheli suite can connect to the board but any attempt to check the escs or read the setup causes esc#1 to boot loop until all power is disconnected. Running the command 1wire 2 from CLI causes esc#2 to bootloop. Here is a video of what happens.

https://youtu.be/x_VOd1nmh1w

Not sure if this is a bug with cleanflight or blheli but I figured id post this.

joshuabardwell, 2015-12-16:
This is most likely not a bug. In most cases where passthrough doesn't work, it's related to the hardware design of the board.

joshuabardwell, 2015-12-16:
Also, the 1wire command has been deprecated. It is no longer in CleanFlight. You just use BLHeliSuite to connect directly to the ESCs in "multi' mode without entering any command in the CLI>

Bravo13, 2016-01-26:
Same trouble with CC3D board

Jahnkeanater, 2016-02-17:
BlHeli does not detect escs on the sparky board. This is an issue with sparky not the escs. I have tested the escs with a naze board and BlHeli connected in multi mode fine.

Thor368, 2016-03-04:
I can confirm the behaviour with CF release 1.12.
The flight control hangs up when "checking" the motor controlers in BLheliSuite in multi mode. I am using a Sparky v1.1 and BLheliSuit v14.1.0.3.1

The generated error message looks like this:
[image]

This might have to do with the VCP. I'm not sure about the implementation of 1wire. Maybe the software emulated serial port behaves different from the hardware silabs bridge.

I will check this by using a hardware UART of the F3 as MSP.

nathantsoi, 2016-03-04:
there is no vcp support for passthrough, not yet anyway

Thor368, 2016-03-05:
Why is that and what needs to be done?

nathantsoi, 2016-03-08:
http://www.rcgroups.com/forums/showthread.php?t=2136895&page=511&highlight=vcp#post34158279

4712, 2016-03-10:
I have taken essential steps in the right direction.
With the help of the implementation of usbVcpGetBaudRate by @tracernz
...
https://github.com/tracernz/cleanflight/blob/vcp-flash/src/main/drivers/serial_usb_vcp.c#L153
https://github.com/tracernz/cleanflight/blob/vcp-flash/src/main/vcp/usb_prop.c#L400

... I finally got the small path-through working (the 4way interface was to memory intensive ~2.8kb), although some testing is to be done. I compile CC3D CF consuming 404 (516 for non reboot version) bytes compared to 312 bytes for the non vcp path-through.

But I noticed that the CC3D:
- counts only 3 of 4 ESC channels
- board and BLHeliSuite hangs when rebooting the board, until the VCP is diconnected.
Betaflight seems not to have this issues.

closed 2017-02-25 by hydra