cleanflight/cleanflight #1587: Serial passthrough for minmosd

Jahnkeanater, 2015-12-16:
Not and issue but idk where to post ideas for features.
There is already gps pass through and 1wire so why not add a command to pass through the serial data to connect to a minmosd through the FC. This would make customizing the osd easier and remove the need for any programmer to set things up.

krzysztofmatula, 2015-12-16:
Not sure about the minimosd gui, but for programming the DTR line is also necessary as far as I know...

JacobGrigg, 2015-12-16:
I believe the DTR line is only required for firmware updates and not configured changes. Alternatively if the minimosd has a reset button it can be used instead of the DTR line.

flounderscore, 2015-12-16:
+1

Even to just change settings that can't be accessed using the odd menu. If DTR is required for reset, power cycling should do the trick as well.

dkisselev, 2015-12-16:
This is a duplicate of another issue, and I believe someone got the changeover 90% implemented, just never opened a PR.

huston007, 2015-12-17:
Anyway please please let's make this done. I'm going to bury my minim osd inside frame and I don't want to disassembly the frame every time I want to configure it

Arakon, 2015-12-20:
Actually, DTR may be skippable if the MWOSD author would implement a reset code that could be sent over MSP, similar to how configurator can send a reset code to cleanflight to get into flash mode.
I'm sure he'd do it.

btimby, 2016-01-15:
I published a release including this feature for testing:

https://github.com/btimby/cleanflight/releases/tag/v.11.0-SerialPassthrough

Arakon, 2016-01-15:
Works for me to configure a micro minimosd on a CC3D nano.
- CP2102 via Mainport
- OSD on Flexport
- serialpassthrough 2 115200 rxtx
Disconnecting, then connecting to the com port of the CP2102 via the MWOSD gui works.

Flashing via Arduino does not, but I believe that is because it's attempting to access the bootloader via 57600 baud:
Using Programmer              : arduino
         Overriding Baud Rate          : 57600
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xb9
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0b
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xad
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xb9
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x0b
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xad
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xb9
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0b
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xad
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xb9

avrdude done.  Thank you.

I didn't actually try with 57600 baud to see if that will allow programming, though.

btimby, 2016-01-16:
I have not been able to get flashing to work yet. There are two main issues:

1. You have to reset the OSD before an upload can proceed. This is usually done by AVRDude using DTR, which is not connected to the FCB.
2. The timing is critical as the bootloader waits only a short time to receive the upload.

I will be working on something, but I don't have an easy way of flashing yet, so this is config-only for now.

btimby, 2016-01-16:
This issue is a duplicate of #1439

DerDominik, 2016-02-09:
Tested it today with a SPRacingF3 and a MicroMinimOSD ("serialpassthrough 2") and seems to work without any failures after closing cleanflight GUI and starting MWOSD-GUI.
Great Work!! Hope it will be integrated in betaflight or cleanflight next time

closed 2016-10-31 by hydra

hydra, 2016-11-01:
done.

PR #1620 (2016-10-31): Implementation of serial passthrough.
Meant to be used for accessing the OSD without removing it.

I folded the gps passthrough functionality into this as the two are similar.

I need some review and comments from a more experienced developer. I am not sure I grok the serial system, mostly I have questions about config and how one goes about locating available serial ports etc. Pointers to example code are welcome.

My attempt to fix #1587