cleanflight/cleanflight #1589: Can't Set IBUS in CLI

qfather, 2015-12-16:
CF 1.12 build #2545
Configurator 1.2.0-dev

I can't set IBUS in the CLI:

    # set serialrx_provider = IBUS
    Invalid value

and

    # set serialrx_provider = 7
    Invalid value

It does work setting it in the GUI.  However, once set in the GUI the CLI reports the following

    # get serialrx_provider
    serialrx_provider = MW23

Pierre-A, 2015-12-17:
**@qfather**, you can't set IBUS in CLI because the `lookupTableSerialRX  (in `io/serial_cli.c`) has not been updated while merging PR #983

huston007, 2015-12-17:
@qfather @Pierre-A could iBus be set by GUI now? Or it totally can't be used?

qfather, 2015-12-17:
@huston007 You can set it in the GUI, but it requires development versions of the Configurator and firmware.  I tested it with Cleanflight build 2545 and Configurator 1.2.0-dev branch.

Pierre-A, 2015-12-18:
This has been fixed in PR #1592, so this issue can be closed.

closed 2016-01-12 by hydra