cleanflight/cleanflight #1598: 1.11.0: Incorrect calculation of altitude when transitioning from sonar to baro

flyinghead, 2015-12-18:
It looks like the recent change to support more sonar hardware (849797a03576d209746a6d8195a8b8c8c6b4a084) has introduced a bug in the altitude calculation.
See in altitudehold.c in calculateEstimatedAltitude():
    sonarTransition = (sonarMaxAltWithTiltCm - sonarAlt) / 100.0f;
sonarTransition is supposed to be between 0 and 1. This is no longer the case, now that the sonar range limits are not fixed.

ledvinap, 2015-12-18:
`sonarTransition = (float)(sonarMaxAltWithTiltCm - sonarAlt) / (sonarMaxAltWithTiltCm - sonarCfAltCm);`

flyinghead, 2015-12-18:
Yep, this looks much better

martinbudden, 2015-12-26:
The change didn't actually introduce a bug, rather it revealed a bug that was already there. Making errors more obvious is one of the reasons for getting rid of magic numbers. Here's the old code:

    if (sonarAlt > 0 && sonarAlt < 200) {
        baroAlt_offset = BaroAlt - sonarAlt;
        BaroAlt = sonarAlt;
    } else {
        BaroAlt -= baroAlt_offset;
        if (sonarAlt > 0) {
            sonarTransition = (300 - sonarAlt) / 100.0f;
            BaroAlt = sonarAlt * sonarTransition + BaroAlt * (1.0f - sonarTransition);
        }
    }

As you can see, the bug was already there.

It still needs to be fixed though! I'll put in a PR.

ledvinap, 2015-12-26:
@martinbudden : Old code did work - sonarAlt was from 200 to 300, so the result was correct..

martinbudden, 2015-12-26:
@ledvinap: yes, you're right 100 = 300 - 200, so the old code did work.

hydra, 2015-12-30:
fix merged.

closed 2015-12-30 by hydra