cleanflight/cleanflight #1599: SonarUnitTest, TestAltitude fails.

AndersHoglund, 2015-12-18:
A small comparision error in sonar.c causes one sonar unit test to fail. The classic "<=" or "<" borderline problem. Either fix the product code:

    diff --git a/src/main/sensors/sonar.c b/src/main/sensors/sonar.c
    index 35f93f3..4d6b8e9 100644
    --- a/src/main/sensors/sonar.c
    +++ b/src/main/sensors/sonar.c
    @@ -147,7 +147,7 @@ int32_t sonarRead(void)
    int32_t sonarCalculateAltitude(int32_t sonarDistance, float cosTiltAngle)
     {
         // calculate sonar altitude only if the ground is in the sonar cone
    -    if (cosTiltAngle <= sonarMaxTiltCos)
    +    if (cosTiltAngle < sonarMaxTiltCos)
             calculatedAltitude = SONAR_OUT_OF_RANGE;
         else
             // altitude = distance * cos(tiltAngle), use approximation

Or the test:

    diff --git a/src/test/unit/sonar_unittest.cc b/src/test/unit/sonar_unittest.cc
    index 5e2ee74..acdc91e 100644
    --- a/src/test/unit/sonar_unittest.cc
    +++ b/src/test/unit/sonar_unittest.cc
    @@ -103,7 +103,7 @@ TEST(SonarUnittest, TestAltitude)
         EXPECT_EQ(sonarCalculateAltitude(400, cosf(DECIDEGREES_TO_RADIANS(224))), 369);
         EXPECT_EQ(sonarGetLatestAltitude(), 369);
         // max range, max tilt
    -    EXPECT_EQ(sonarCalculateAltitude(sonarMaxRangeCm,  cosf(DECIDEGREES_TO_RADIANS(sonarMaxTiltDeciDegrees))), sonarMaxAltWithTiltCm);
    +    EXPECT_EQ(sonarCalculateAltitude(sonarMaxRangeCm, cosf(DECIDEGREES_TO_RADIANS((sonarMaxTiltDeciDegrees -1)))), sonarMaxAltWithTiltCm);
         EXPECT_EQ(sonarGetLatestAltitude(), sonarMaxAltWithTiltCm);
     }

I think the correct way is to update the product code, since there are other tests doing range tests and seems to expect the "<" comparision. And there are already a test doing the "-1" thing, but with hardcoded values. No big deal, either way is fine and gets the UnitTest to PASS:
/A

AndersHoglund, 2015-12-18:
Perhaps need to add that the comparsion is made on floats. Comparing equality on floats is dangerous. Using "<=" is of course less dangerous, but there is still an equlity element that is hard to define with floats. Using "<" or ">" are much safer and less ambiguous.

martinbudden, 2015-12-19:
I've created a pull request to fix this:

https://github.com/cleanflight/cleanflight/pull/1601

The problem arose because previously angles were compared, but after https://github.com/cleanflight/cleanflight/pull/1581 cosines of angles are compared, so the edge cases are slightly different. The test code failed where the edge cases are tested.

AndersHoglund, 2015-12-19:
Strange thing is that the UnitTest seems to work on the Travis CI machine, but fails on my Linux Jenkins server. Compiler/Environment differencies in handling floating point equality comparisions when testing at edge values, I suspect. Best would be to remove the equality part for robustness ....

AndersHoglund, 2015-12-19:
Forgot to say thnx for the PR.
Should have done that myself, but I have betaflight forked in my github account and github will not let me also fork cleanflight. Because of the same origin I guess.
I could have created a new github account of course....

martinbudden, 2015-12-19:
Yes, the test code probably needs an update to better handle the edge cases. There wasn't a problem when the angles in deciDegrees were being used, but now there is a floating point comparison of the cosine of the angle...

On my todo list to update the test code around this area.

hydra, 2015-12-25:
@AndersHoglund just add cleanflight as a remote to your betaflight local repo and create branches like 'cleanflight-master' and push that to github account, then you can also create branches from that for PR's.

@martinbudden Thanks for the PR.  Will merge now.

hydra, 2015-12-25:
#1612 is now merged.

closed 2015-12-25 by hydra

AndersHoglund, 2015-12-25:
Well, I tried forking cleanflight in github. But since I already have betaflight forked, github will not let me. I also already have betaflight as my remote repo in my local repo. I think I might need two github acounts if I should follow/fork both cleanflight and betaflight.

hydra, 2015-12-25:
you don't need two accounts.  Just add the cleanflight code to your betaflight fork and push the branches to your betaflight fork using different branch names locally and on the remote.