cleanflight/cleanflight #1602: Change Request: ESC throttle parameter changes.

AndersHoglund, 2015-12-19:
There is a lot of confusion on what the current ESC throttle parameters are actually doing. I would like to propose new more intuitive names :

    full_throttle    (Replacing max_throttle )
    idle_throttle    (Replacing min_throttle )
    stop_throttle    (Replacing min_command )

Might be a lot of work for little or no functional gain. But I think user confusion and the number of support cases would be less.
/A

ProDrone, 2015-12-19:
I was thinking along the same lines, but the problems i see are:

1. Changing parameter names may cause more confusion because they were used for so long and there are many online references and videos that use them.
2. Choosing new names without any ambiguity in the used terms. For instance the `throttle` part in your proposal is still confusing (throttle is associated with the input side). Maybe something like:
 ~~~
max_motor
idle_motor
stop_motor
~~~
Maybe we need to vote for proposed names?

3. There are a lot more candidates once we start changing parameter names... :/

AndersHoglund, 2015-12-19:
1. Yepp. Hard to change something that has spread too far and wide
2. Agree, "throttle" should probably be avoided in this end. 
3. Really ? ;-)

borisbstyle, 2015-12-21:
+1

I totally agree on this one. It is becoming the most annoying question all around forums.

borisbstyle, 2015-12-21:
The most confusing is the min_check and max_check as well.

I like these:
max_motor
idle_motor
stop_motor

min_check = min_rc_check
max_check = max_rc_check

hydra, 2016-01-08:
@borisbstyle +1 I find the word 'check' in the above unhelpful, always have.

Marking this as low priority for now.  I think when we remove the CLI things will be easier since we can just update the GUI instead.  See #922.  Blocking.

waltr1, 2016-03-27:
I like these also:
max_motor
idle_motor
stop_motor

min_check = min_rc_check
max_check = max_rc_check

All five of these has and continues to cause much confusion.Boris' names are much clearer.

Teracis, 2016-03-27:
I like @borisbstyle suggestion of:
I like these:
max_motor
idle_motor
stop_motor

min_check = min_rc_check
max_check = max_rc_check

Except with one change that it's min_motor instead of stop_motor just for 3D purposes because in 3D mode min_command is actually full negative thrust / full reverse motors so it's definitely not motors stopped.

I'll leave it up to you lot though, I can get used to any term you decide to make it.

MasterZap, 2016-03-27:
Agree with Dominic, "check" says nothing. I'd rather see something way more explicit:
low_stick_command
high_stick_command

Because even "min" and "max" is wrong here. min_check is not a mininum, it is actually the MAXIMUM level UNDER which the stick can be used as a command. And max_check is not a maximum, it is actually the MINIMUM level OVER which there can be a stick command....

MasterZap, 2016-03-27:
Or even super explicit:

stick_command_low_boundary
stick_command_high_boundary

Yet yet YET another option is of course to tell people to calibrate to 1000-2000 and just implicitly set these internaly to the last 3% of range in either direction....

fftunes, 2016-03-27:
ESC_max   or  motor_max
ESC_min   or  motor_min   (not "stop", since it's used for 3D too)
ESC_idle  or  motor_idle

rc_command_low  or  rc_check_low   and  .._high

AndersHoglund, 2016-03-27:
min_check and max_check are not any "checks". Not "commands" nor "boundaries" either.
They are threshold values. So, my suggestion on these are:

    rc_low_threshold
    rc_high_threshold

Or maybe:

    stick_low_threshold
    stick_high_threshold

MasterZap, 2016-03-28:
Rather than calling "stop" "min" because "it's used in 3D", then rather don't use it in 3d. Calling it ESC_Stop or Motor_Stop would clean up SOOOO much confusion.

If that means adding an ESC_reverse_max then so be it.

MasterZap, 2016-03-28:
They may be "thresholds" but they are thresholds used FOR the stick commands.

AndersHoglund, 2016-03-29:
Yeah, stick commands are one thing, coming from Tx/Rx. Stick thresholds another, in the FC. 
A speed limit sign on the road side is not a speed command. Passing over this threshold will/might get expensive.
/A

closed 2017-02-25 by hydra