cleanflight/cleanflight #1609: 1.10-1.11, Failsafe w/OneShot will not disarm motors completely

jlong23, 2015-12-21:
Configuration: Flip32 with Fail-safe & One-shot enabled on EMAX BLHelis, Arm in any mode, enable Fail-safe via Transmitter Failure or Fail-safe Mode Switch.   The Motors will slow, but will shut off, some will react in unpredictable behavior (Like they are getting false pulses).  

Turn off One-Shot and repeat Fail-Safe procedure and the motors disarm as expected.

Otherwise, sweet work guys!

timdonovanuk, 2015-12-21:
Hmm I just posted something similar here (last comment): https://github.com/cleanflight/cleanflight/issues/912

I've not tried toggling OneShot off (I have it on). Will try this week.

Have you tried enabling MOTOR_STOP (just to test)?

jlong23, 2015-12-22:
Sorry, yes.

When OneShot: Off, MotorStop: On, FailSafe: On seems to work as expected for failsafe.  I should have done a CLI dump and perhaps a YouTube with the demo behaviour.

borisbstyle, 2015-12-22:
Did you try lowering your failsafe throttle and min_command?

Are you also using emf_avoidance?

ukscaleandcrawler, 2016-01-01:
i am having the same problem had to go back to libra pilot really wanna use betaflight also i noticed that i carnt calibrate the escs either and its because i carnt change the board loop time to 2500

hydra, 2016-01-12:
this seems like a support issue.  no response from the user after 20 days. closing.

closed 2016-01-12 by hydra