cleanflight/cleanflight #1610: suggestion: better layout for small screens please!

huibuuh, 2015-12-21:
I noticed today after my cleanflight updated itself, that it is now almost unusable on my small screen. In the pre 1.x versions I could easily use cleanflight on my small screen in the field. Now I can hardly see things anymore.

please change the layout so that it is possible to use the program on small devices outside in the field again!
[image]

joshuabardwell, 2015-12-21:
The configurator is a separate project, and is where this issue should be raised.

https://github.com/cleanflight/cleanflight-configurator

SteveAmor, 2015-12-21:
Please can you close this issue and open one under the cleanflight configurator repo.

closed 2015-12-21 by huibuuh