cleanflight/cleanflight #1611: GPSHold and GPSRTL with CC3D v1.11.0

mastersurferde, 2015-12-21:
Hello Developer and Friends of Cleanflight.
I am a new user of CC3D with Cleanflight. Since 2 Weeks I try to get the GPS Fight modes working. But with no Success.
I added a Barometer - ok working
I added a Compass - ok working
I added a GPS-Module - it is working and I get a 3d Fix.
I am able to choose the GPSHold and the GPSRTH - but the GPS is not correcting the flight direction.

Is the CC3D able to work with the GPS-Modes? Or do I need a Naze32?
I can´t find any Documentation about this.

Thank you
Stefan

stronnag, 2015-12-21:
This is not an issue(bug),  it is a support request. Please ask on RCG or IRC.

mastersurferde, 2015-12-21:
ok. Sorry.

SteveAmor, 2015-12-21:
@mastersurferde can you please close as it's a support request.

closed 2015-12-21 by mastersurferde