cleanflight/cleanflight #1616: 1.11 led strip and servo tilt/servo mapping causes crash flip32

astragalizont, 2015-12-22:
I wanted to activate ledstrip and servo tilt plus servo mapping on a quad X configuration, so i can use a switch with PPM. 

With servo tilt and servo mapping the Motors shift to 3-6 and output channel 1+2 can be used for servo/switch etc.

Led strips work too, with no problems.

But if i activate both features at the same time, the board crashes and can't boot anymore. Had to reflash with bootpad shorted. Normal flashing didn't work either.

hydra, 2015-12-22:
please can you provide a configuration dump

When does the crash occur anyway?
a) After saving the configuration or b) when activating something?
if b) tell me what the RX channel values are need to cause the crash.

astragalizont, 2015-12-23:
Hello, thanks for your answer. 

Crash occurs when i hit save and reboot. 

How do i dump the configuration? 
Cli dump or cleanflight backup? 

With regards

astragalizont, 2016-02-03:
I checked it with version 1.12, the Problem still persists.
If led strip and channel forwarding are saved in the GUI, the board doesnt crash, but only channel forwarding is active, led strip is switched back to grey "off"

Servo tilt and channel forwarding and servo tilt together work.

Servo tilt and led strip  can't be used together, servo tilt is active but led strip is still off.

I startet with a clean flash in cleanflight and checked wich additional setting will crash the board and it is:

PPM with oneshot, ledstrip, servotilt and channelforwarding.

If i activate these settings in the GUI and hit save the GUI tells me its isrebooting, but cant reconnect after that. And if i connect and disconnect again this: No configuration received within 10 seconds, communication failed

regnets, 2016-06-01:
I got the same problem with Cleanflight 1.12.1 and a NAZE32 V5.

I would like to use the lost plane finder on the matek v3 pdb.

I also got a WS2812 LED Strip. If i activate Servo tilt AND the LED feature Cleanflight isn't booting anymore.

astragalizont, 2016-06-06:
I can't post a dump, because the FC crashes and can only be revived via reflash and connecting the bootpins.

blinkin69, 2016-09-19:
i have the same problem with the latest cleanflight and NAZE32 REV 6. Can you fix it that we can use servo tilt(used for matek 5in1 buzzer) and LED together. Thank you...

closed 2017-02-25 by hydra