cleanflight/cleanflight #1621: Merge iNav GPS navigation code with Cleanflight

digitalentity, 2015-12-23:
Navigation (iNav fork) is reaching stable state, soon we could be starting to merge it to CF.
There is, however a big problem - memory. iNav barely fits into F1-based targets. I al already disabling eye-candy reafures - ledstrip, display. F3 targets seem to be capable enough to have everything.

GPS/Navigation stuff is taking ~29K at the moment.

There are some options:  
1. @marksev1 suggested that we have different "flavors" of Cleanflight - racing, focused on lowest possible looptimes, and navigation, focused on best navigation performance.
1. Drop the CLI and create an extension to MSP to be able to access parameters in the similar way we do with CLI, but without the command-line parser. CLI takes about 24K flash, if we drop it we will have plenty of space for features to improve flight performance.
1. Split multicopter and airplane firmwares. This will greatly reduce the memory requirements for navigation features (probably by 7-10K)
1. Remove some other code. One of the candidates is NMEA GPS code - does not give all of the required data (velocity in 3D space, DOP is measured in arbitrary units). Binary protocols are more efficient and easier to parse.
1. Disable navigation for F1 targets completely.
 
I'm opening this issue to discuss it with the community.

skaman82, 2015-12-23:
I would suggest to drop the CLI. Even when it means more work. I think that removing Navigation from F1 targets is a bad idea.

digitalentity, 2015-12-23:
@skaman82 I agree. Personally, I feel that F1's are going to be around for a several years. I also very much like performance of navigation on NAZE/CC3D (no surprise at it was developed and tested on an F1 board) :smiley:

marksev1, 2015-12-23:
+1 F1 4life :)...No. 5 is out of the question :).

Seems that reworking cli is the obvious first candidate. But would need to maintain same user experience in the configurators as to avoid confusion among users.

Zero9r, 2015-12-23:
F1 Targets need to have this.  I say if its space get ride of the NMEA GPS Code if it is not needed with this new Nav.  Also I don't think two versions for multi and airplane is a bad idea either.  Makes it a cleaner flash I would think.

What I would like to see for my new 450 build is Navigation, LED Lights, and the stability that Boris;s Beta Flight gives.  CLI can be removed if the GUI can do everything I believe too.  why not make it a simple clean build with only those features.

JacobGrigg, 2015-12-23:
Would having seperate versions for multi and airplane affect those who want to built VTOL craft?

digitalentity, 2015-12-23:
@JacobGrigg 
Yes, having two separate versions would probably affect future development of VTOL support.

@Zero9r 
Stability features from Betaflight are being merged into CF. Some are already there (gyro sync, FIR filtering). I think the best way would be to remove CLI and make the configurator access all parameters via MSP. If that wouldn't be enough - drop NMEA support.

Maybe somebody would come up with more ideas on how to fit everything in?

Zero9r, 2015-12-24:
I think that Idea is just fine. I have no issues using MSP only

Linjieqiang, 2015-12-24:
Why not thinking support for F4 board Such as Sparky v2 ? But I think remove CLI is a best way .

digitalentity, 2015-12-24:
@Linjieqiang 
Eventually we'll have F4 board support. OP Revo is already supported in Sambas's fork. But limiting navigation features to more powerfull boards only is a bad idea. Naze/CC3D boards can go navigation with great performance, and there are still lots of them around, available and dirt cheap.

Linjieqiang, 2015-12-24:
@digitalentity
Oh,I see.Thanks for your work!

JohnOCFII, 2015-12-24:
> 1. Have different "flavors" of Cleanflight - racing, focused on lowest possible looptimes, and navigation, focused on best navigation performance.
> 2. Drop the CLI and create an extension to MSP to be able to access parameters in the similar way we do with CLI, but without the command-line parser. 
> 3. Split multicopter and airplane firmwares. 
> 4. Remove some other code. One of the candidates is NMEA GPS code 
> 5. Disable navigation for F1 targets completely.

In order of preference, I think item 2 (drop CLI), followed by item 4 (remove NMEA GPS code) would be the best place to start.  If more space is needed, I'd like to see some discussion of on-the-fly compile/download options, similar to the OpenTX model, where the configurator lets you choose what features you want when you download new firmware.  If this is too complex or difficult to develop or maintain, then we look elsewhere.

Ravid824, 2015-12-24:
We still want F1 support. Same as others, Drop the CLI and create an extension to MSP is the best choose for me. It can be done together with removing NMEA GPS code to release more space for future developments.

digitalentity, 2015-12-24:
Seems that everybody is ok with removing the cli. @hydra what do you think?

JohnOCFII, 2015-12-24:
> Seems that everybody is ok with removing the cli. @hydra what do you think?

I think he is in favor of removing the CLI, given this previous issue:  https://github.com/cleanflight/cleanflight/issues/922

skaman82, 2016-01-08:
Working on this in here: https://github.com/skaman82/cleanflight-configurator/issues/59

andre92, 2016-05-25:
> 1. Have different "flavors" of Cleanflight - racing, focused on lowest possible looptimes, and navigation, focused on best navigation performance.
> 2. Drop the CLI and create an extension to MSP to [...]

for item 4, how much space can be saved by leaving out pid controllers LUX and MW2.3 on F1 targets?

skaman82, 2016-05-25:
I vote NO to "Disable navigation for F1 targets completely"

TinoGroteloh, 2016-06-29:
Since I am really waiting for this issue to be solved, I set up a little [bounty](https://www.bountysource.com/issues/29355582-merge-inav-gps-navigation-code-with-cleanflight). Consider doing the same if you are in the same situation as I am :)

DieHertz, 2016-07-27:
What's the current status on this one? It would be nice if you could provide some directions.
As far as I've understood, the idea is to remove CLI and replace it with a binary sub-protocol injected into MSP, right?

martinbudden, 2016-07-27:
The plan is to eventually merge the iNav navigation code into Cleanflight. However there are a lot of others issues being worked on at the moment.

There has also been some divergence between the iNav and Cleanflight codebases, in particular in the area of IO and parameter groups - these areas are probably best converged again before the iNav code is moved over (that is iNav to adopt parameter groups and Cleanflight to adopt the new IO). These changes are planned and/or occurring.

The upshot of all that is that this is not going to happen soon.

strips, 2016-08-07:
Personally I think CLI makes things simpler. Not everything fits in a GUI. I'm probably a bit biased as I prefer to do anything in CLI on computers...

But I think todays (not yet made) decision to drop CLI and make it a point to enable on F1 at all costs might bite us back in the ass sooner than later.

Yes it makes sense but then there is so many cool features for navigation that eventually F1 support will just be a limiting factor. 

I can support dropping CLI but I hope it will not limit future development. Like better algorithms, better filtering and advanced navigation features. Some times you just got to get rid of the old and be tough enough to embrace the new. This is a difficult balance in all aspect of life.

closed 2017-02-25 by hydra

Ravid824, 2017-02-25:
Issue #197 (Flight - Improve GPS Position Hold (GPS PH). Closed this issue #1621 as well saying that the plans to merge the iNAV code dropped?
We are waiting for long time for this to happen , very disappointing for many of the cleanflight users.
Any chances that this will happen sometime soon ??????????

At the minimum, is #2521 (FC lock due to UBLOX M8N overload?) is going to be fixed
hydra  mentioned "the code in iNAV which addresses this will be merged in due course"

Tinoooo, 2017-02-25:
@hydra Why did you close the Issue?

andre92, 2017-02-26:
@Tinoooo he closed all issues at once, probably because of the high number of invalid issues (for newbie support we have RC Groups, this place is only an issue tracker to report legitimate bugs or request features)
@Ravid824 iNav has surpassed Cleanflight in many aspects. Not sure if it is ever going to be merged but at this point you might as well just use iNav

McGiverGim, 2017-02-26:
I think all the issues were closed because version 2.0 will be based in
betaflight, another new start...