cleanflight/cleanflight #1625: Error in normalising DTerm to cycle time in pidLuxFloat

martinbudden, 2015-12-26:
In `pidLuxFloat` the value of `DTerm` is normalised to the cycle time by the code:

        //-----calculate D-term
        delta = RateError - lastError[axis];
        lastError[axis] = RateError;
        // Correct difference by cycle time. Cycle time is jittery (can be different 2 times), so calculated difference
        // would be scaled by different dt each time. Division by dT fixes that.
        delta *= (1.0f / dT);

However `pidLuxFloat` is called in the order of 1000 times per second, so `dT` is of the order of 0.001. This means, as part of the normalisation, `delta` is multiplied by approximately 1000. This means that when `DTerm` is contrained by:

            DTerm = constrainf(deltaSum * pidProfile->D_f[axis] * PIDweight[axis] / 100, -300.0f, 300.0f);

the constraints are used practically all the time.

Now this would not actually be a problem if sufficiently small values of `pidProfile->D_f` were chosen, but in practice this is not the case. For example the default values of D set in `config.c` are:

    pidProfile->D_f[ROLL] = 0.03f;
    pidProfile->D_f[PITCH] = 0.03f;
    pidProfile->D_f[YAW] = 0.01f;

These values are all about 10 times too large, since they are all effectively multiplied by 1000 in `pidLuxFloat`.

This error was revealed by the new PID test code, see:

https://github.com/cleanflight/cleanflight/pull/1624/files#diff-a52cad50ebd611ad01aa1d7c4fc1c70eR448

the failing test is commented out, ready to be reinstated once there is a fix. Note that the constraints are hit for a relatively small value of `rateError`.

It should also be possible to see this error manifest itself using blackboxing. It should be seen in the values of `axisPID_D[axis]`: they will have a three state distribution, that is they will either be -300, +300 or a value close to zero.

There are two ways to solve this problem:

1. Either set the default `pidProfile->D` values to be about 10 times smaller
2. Or change the normalisation so that such small values of `pidProfile->D` are not required (and again change the default `pidProfile->D` values to match the new normalisation.

In either case users will have to change their manually tuned P values.

I favour the second approach, namely

    delta *= (1000.0f / dT); // normalise delta to 1000Hz cycle time

and then change the defaults:

    pidProfile->D_f[ROLL] = 3.0f;
    pidProfile->D_f[PITCH] = 3.0f;
    pidProfile->D_f[YAW] = 1.0f;

Of course everyone who has manually tuned with have to change their D values, but I think in many cases they were not working proberly anyway.

ledvinap, 2015-12-26:
`delta *= (0.001f / dT); // normalise delta to 1000Hz cycle time`

digitalentity, 2015-12-26:
D term is a derivative. Derivative is calculated by dividing the change in
the value by the time over which this change happened. If you sample two
times faster you'll haggle have two times smaller interval but also two
times (approximately) smaller change.
Lux works exactly like that, it's mathematically correct way.

martinbudden, 2015-12-26:
@ledvinap well spotted. You comment is of course correct. Oops on my part.

martinbudden, 2015-12-26:
@digitalentity , yes I understand DTerm is a derivative, ie, in essence:

    DTerm = pidProfile->D_f * (rateError - lastError) / dT

additionally we use a moving average and/or some filtering to remove high frequency noise and then we apply a constraint to make sure the `DTerm` is not too large.

My argument is that the  `pidProfile->D_f` values have to be very small to avoid the contraint coming into effect. Currently the default values of `pidProfile->D_f` are too large.

Note that the `pidMultiWiiRewrite` PID (which is essentially an integer version of `pidLuxFloat`) does not apply a contraint to the `DTerm`, since, presumably, it does not have this scaling problem.

digitalentity, 2015-12-26:
Then there's no error in DTerm calculation and the whole issue is titled
wrongly.
What you are really trying to do is to scale up the D_f to get better
resolution for tuning, right?

borisbstyle, 2015-12-26:
I think the numbers in luxfloat are in general too narrow. Configurator and other tools are not helping either as no floats can be configured there.
My 4S machines literally have just 5-10 tuning steps in the most common tools, unless I use cli.

It's a good point to scale D differently as it is way too aggressive

borisbstyle, 2015-12-26:
BTW value of 3.0f is not supported in configurator.

MSP should be using higher range for luxfloat

digitalentity, 2015-12-26:
I guess it would be better to modify msp that sets D_f value (scale integer
received from configurator to float), or soon enough we'll get another
issue like "fix incorrect derivative calculation in lux".

borisbstyle, 2015-12-26:
@digitalentity 
I already did a PR for that long time ago, but got lost somewhere. I even think I deleted it.

borisbstyle, 2015-12-26:
@digitalentity 
I believe the configurator work was messy to get it done due to the current structure of those PID pages

digitalentity, 2015-12-26:
No need to change configurator, change Cleanflight msp handling code. And
let configurator still be integer based

borisbstyle, 2015-12-26:
No it cant. The resolution in the configurator is too low

borisbstyle, 2015-12-26:
It's using 8bit integer now, which basically means 255 steps.

Besides that configurator offers only 1 decimal for P and none for D. So if you want the real values you def need to change stuff in configurator.

borisbstyle, 2015-12-26:
We can also scale the numbers in luxfloat as well. Just make the tuning range higher.

digitalentity, 2015-12-26:
You are right about the configurator and tuning range. But I think it would
be bad idea to mess with calculation of derivative. LUX is the only
controller that is coded according to calculus, let's not break it.
I think we can safely remove the constraint for DTerm in lux. My guess is
that it is there to deal with derivative kick which is better handled by
median filtering.

borisbstyle, 2015-12-26:
@digitalentity 
Yeah I am working on Dterm lately as you know :D. Median filter certainly is better than the moving average.
But I still think that D might be downscaled a bit in luxfloat. Probably the normalizing part.

This will make it 10 times smaller and give the same effect with the current pids
delta *= (0.1f / dT);

ledvinap, 2015-12-26:
@borisbstyle : Median filter is nonlinear ... it may cause unexpected problems ...

digitalentity, 2015-12-26:
@borisbstyle, @martinbudden 
I suggest leaving delta calculation as it is to keep derivative calculation straight-forward and easy to unerstand:
```
delta *= (1.0f / dT);
```
And change scaling where DTerm is actually calculated:
```
deltaSum * pidProfile->D_f[axis] * PIDweight[axis] / 100
```

ledvinap, 2015-12-26:
Maybe 16bit floats (https://en.wikipedia.org/wiki/Half-precision_floating-point_format) for MSP messages is way to go.
The range is enough to represent 16bit value for integer based PIDs (and other calculations) and resolution is (probably) good enough for float based ones.

digitalentity, 2015-12-27:
@ledvinap 
Or a fixed-point in Q6.10 format, that will give us range from -32 to 32 and resolution of ~0.001

borisbstyle, 2015-12-27:
Still the fix in the configurator would be dirty on the way how the old one was done :D. I don't know how it would fit in the new configurator. Maybe it is easier there.

borisbstyle, 2015-12-27:
@ledvinap
median filter is better IMO than just a moving average there is now. Not
sure how you expect problems compared to the moving average?

I will be experimenting with it.

martinbudden, 2015-12-27:
@digitalentity: in retrospect the term "error" was a bit strong, my apologies. In fairness though, I did not say there was an error in the calcualtion of the DTerm (nor did I wish to imply this), merely that there was an error in its normalisation.

"What you are really trying to do is to scale up the D_f to get better resolution for tuning, right?" - yes, that is one of two things I am trying to do. See also #1626 . If we address this and #1626 then all PID values would be in the range 1 to 10, making for an easier tuning experience.

The other is to ensure that the constraint on the DTerm only comes into effect rarely. As I said, it is my view that because of the current scaling and values of pidProfile->D_f used, this constraint is invoked far more often than it should be.

"I suggest leaving delta calculation as it is to keep derivative calculation straight-forward and easy to unerstand ... And change scaling where DTerm is actually calculated" - I agree. How about:

    delta = (RateError - lastError[axis]) / dT;
    ...
    DTerm = constrainf(0.001f * deltaSum * pidProfile->D_f[axis] * PIDweight[axis] / 100, -300.0f, 300.0f);

I'm not fixated on the 0.001f scaling - if people prefer 0.01f or some other similar value I'm OK with that. (And I'd eventually like to see the constrainf go at some point.)

"LUX is the only controller that is coded according to calculus, let's not break it." I agree entirely. I've written a bunch of test code to help ensure this, see: https://github.com/cleanflight/cleanflight/pull/1624 
Indeed, the reason for this issue report was because of what I found when writing and running the test code. (I'm also aware of the plans for a median filter for the derivative and realise the test code will need some minor modifications for some of the tests when this comes into effect.)

digitalentity, 2015-12-27:
I believe that constrain on DTerm is unnecessary. PID output is always constrained, and high DTerm is usually caused by derivative kick (can be filtered out by median) or extreme angular acceleration (is a normal functioning of DTerm).

bdoiron74, 2015-12-27:
Constraints are necessary. Remember that the output get summed up with the other 8 PID terms to create a matrix that is added to the base throttle. Best case, at mid throttle, there are less than 500 'units' available [(max_throttle - min_throttle) / 2].

digitalentity, 2015-12-27:
@bdoiron74
I'm talking about this specific constrain:
```
DTerm = constrainf(deltaSum * pidProfile->D_f[axis] * PIDweight[axis] / 100, -300.0f, 300.0f);
```

BTW, no constrains are used in MultiiWiiRewrite and this does not prevent it from working.

martinbudden, 2015-12-27:
For pidLuxFloat there is an overall constraint, namely:

    axisPID[axis] = constrain(lrintf(PTerm + ITerm + DTerm), -1000, 1000);

I agree with digitalentity that the individual contraint on the DTerm is probably unnecessary.

bdoiron74, 2015-12-27:
ok - 'necessary' is a strong word, and the overall PID constraint would clip it anyway. Except of course, [-1000,1000] is well beyond the actual available motor output.

ledvinap, 2015-12-27:
@borisbstyle : One problem with median filter is that it may change mean value of signal.
Extreme example is sine samples at fs=3f (http://www.gaussianwaves.com/gaussianwaves/wp-content/uploads/2014/07/Sampling_Continous_time_signal_in_Matlab.png). Applying 3 sample median will result in constant -0.5 level.

Changed mean of signal will cause problems similar to aliasing ...

Median filter is great to filter random spikes in signal. But gyro/acc noise is quite random (but motor frequency is leaking in). Linear techniques are quite powerful here ...

borisbstyle, 2015-12-27:
@ledvinap 
Thanks for clarification. I didn't look at it that way at all.

borisbstyle, 2015-12-28:
@ledvinap 
You were right. I did side by side comparisation of real blackbox data for delta with moving average and median. Median was indeed showing some "lower" frequency ripples, which were not there with the simple averaging method.

ledvinap, 2015-12-28:
@digitalentity 
> Or a fixed-point in Q6.10 format, that will give us range from -32 to 32 and resolution of ~0.001

16bit floating point will cover most (probably all) parameters used in cleanflight, without need for explicit scaling. It may simplify configurator and code ...

martinbudden, 2015-12-30:
Shall I summarise then:

Currently the cleanflight configurator uses 8 bit values for the PID values (both in the UI and the MSP). This does not really provide sufficient range or resolution for PID tuning. There are already workarounds in the cleanflight code because of this (multiplying the ITerm by 10 https://github.com/cleanflight/cleanflight/pull/666) and boris finds that the D values are currently too aggressively scaled.

To improve PID tuning the configurator should use 16 bits for the PID values, there are two proposals:
ledvinap suggests using 16bit floating point values, digitalentity suggests Q6.16 format (which gives a range of -32 to 32 and a resolution of ~0.001).

Once the configurator has been changed it will be possible to change the scaling of the PID values in `pidLuxFloat` (for example remove the ITerm multiplier of 10, and divide the DTerm by 1000 (or 100)) to bring the PID values into a easy to use range (ie all in the range of about 1 to 10).

Does that sum it up? Or have I misunderstood something?

for reference here's the PID tab HTML

    <td><input type="number" name="p" step="0.1" min="0" max="25.5" /></td>
    <td><input type="number" name="i" step="0.001" min="0" max="0.255" /></td>
    <td><input type="number" name="d" step="1" min="0" max="255" /></td>

the PID tab javascript

    $('.pid_tuning .ROLL input').each(function () {...
    switch (i) {
        case 0:
            $(this).val(PIDs[0][i++].toFixed(1));
            break;
        case 1:
            $(this).val(PIDs[0][i++].toFixed(3));
            break;
        case 2:
            $(this).val(PIDs[0][i++].toFixed(0));
        break;
    }

and the MSP

    process_data: function (code, message_buffer, message_length) {...
        case MSP_codes.MSP_PID:
            PIDs[i][0] = data.getUint8(needle) / 10;
            PIDs[i][1] = data.getUint8(needle + 1) / 1000;
            PIDs[i][2] = data.getUint8(needle + 2);
            break;
    
    MSP.crunch = function (code) {...
        case MSP_codes.MSP_SET_PID:
            buffer.push(Math.round(PIDs[i][0] * 10));
            buffer.push(Math.round(PIDs[i][1] * 1000));
            buffer.push(parseInt(PIDs[i][2]));
            break;

NightHawk32, 2015-12-30:
@martinbudden you got most of it.
But you also have to consider compatibility to old versions.
This consists of three points:
- backup and restore: old backup files which still have the old scaling have to be mirgrated properly
- MSP: this has to check wich api level is connected and therefore decide if 16 or 8 bit PID values are used.
- the frontend: depending on the api level step and max have to be set

Since this would require a api level push, so it makes sense to include this before the next release, because the api level has been pushed for current developments anyway.

martinbudden, 2015-12-30:
@NightHawk32 yes file compatibility has to be considered as well. I've just had a hunt around and found this in `serial_msp.c`:

    case MSP_SET_PID:
        if (IS_PID_CONTROLLER_FP_BASED(currentProfile->pidProfile.pidController)) {
            for (i = 0; i < 3; i++) {
                currentProfile->pidProfile.P_f[i] = (float)read8() / 10.0f;
                currentProfile->pidProfile.I_f[i] = (float)read8() / 100.0f;
                currentProfile->pidProfile.D_f[i] = (float)read8() / 1000.0f;
            }
        break;
    ....
    case MSP_SET_PID_FLOAT:
        for (i = 0; i < 3; i++) {
            currentProfile->pidProfile.P_f[i] = (float)read16() / 1000.0f;
            currentProfile->pidProfile.I_f[i] = (float)read16() / 1000.0f;
            currentProfile->pidProfile.D_f[i] = (float)read16() / 1000.0f;
        }

and there is corresponding `serialize` code. So it looks like this work has been done in cleanflight.

However there is no corresponding code in the configurator. `msp.js` only deals with the `MSP_SET_PID` command, not `MSP_SET_PID_FLOAT`

So it looks like the decision on the decision of the PIDs has been made: 16bit scaled by 1000. It also looks like the file compatibility issue has been addressed.

borisbstyle, 2015-12-30:
You are looking in my branch where I was playing with that :).

Thats not there in cleanflight.

martinbudden, 2015-12-30:
That's a bit embarassing :-) I must have forgotten to checkout master before I started browsing the code. I must admit I was surprised when I found it.

I was thinking a bit more about the roll pitch yaw PID values. Why not just store them as floats? There are only 9 of them (3 PIDs for 3 axes). Storing them as 16 bit values costs 18 bytes, whereas storing them as floats costs 36 bytes. For a cost of 18 bytes we can make them totally future proof. (I'm not one for wasting bytes, but saving these 18 bytes does not seem worth it - neither it terms of storage space or transfer time). The other 7 of the 10 PIDITEMS in the MSP don't need to be stored as floats, they can be stored as 16 bit quantities.

martinbudden, 2016-01-12:
I'm closing this issue. The discussion has revealed that there is an underlying problem: the storage of PIDs as 8 bit values does not allow sufficient range or resolution for proper tuning. I'll raise an issue for that and close this one. (Once the underlying issue is addressed it will be possible to revisit the scaling of the PID values.)

closed 2016-01-12 by martinbudden