cleanflight/cleanflight #1626: Inconsistent normalisation of ITerm in pidLuxFloat

martinbudden, 2015-12-26:
In `pidLuxFloat` the value of `ITerm` is calculated as follows:

        // -----calculate I component.
        errorGyroIf[axis] = constrainf(errorGyroIf[axis] + RateError * dT * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);

Note the multiplication by 10.

This multiplication is unnecessary and I suspect is a hangover from when the code was ported from the integer PID code. Of course if the multiplication by 10 was removed in the PID code then the default `pidProfile->I_f` values would need to multiplied by 10. Additionally users who have manually tuned their PIDs would have to multiply their I values by 10. However #1625 if fixed then users will have to change their D values, so changing their I values at the same time won't be too problematic.

Note that if this change is made along with #1625 then the P, I and D values will all be normalised to the same value, namely 1000Hz. (P values don't actually need normalising). This will mean P, I and D values will all have the same order of magnitude, namely all taking values between 1 and 10.

borisbstyle, 2015-12-28:
I remember when this came. It's from the times when luxfloat was just introduced and folks were still experiencing lack of I even when crancked to the max values in the configurator.
So this was done to increase the configurable range.

I think the real solution would be to fix it in the configurator.

martinbudden, 2015-12-28:
Interesting. So, yes, this should be fixed in the configurator and then we can remove the *10 multiplier.

Also if digitalentity's iNav code is merged then people are likely to start running Cleanflight on much bigger birds, ie 650 class, 800 class or even larger. For these size quadcopters they are going to need the flexibility to set much larger PID values, so the configurator needs to allow them to do this

borisbstyle, 2015-12-28:
See here the original PR:
https://github.com/cleanflight/cleanflight/pull/666

I agree about changing configurator and MSP for floating point pid controllers. A simple multiply and divide by 1000 with 16 bit would give enough resolution.

martinbudden, 2016-01-03:
I'm closing this. It is a workaround because the MSP and the configurator use 8-bit values for the PIDs, and it cannot be addressed until the MSP and the configurator are changed.

closed 2016-01-03 by martinbudden