cleanflight/cleanflight #1628: Issue with Gtune value below 10 in current dev master

droidparanoid, 2015-12-27:
Hi guys,

I found a problem in the current dev master (1.12.0) with Gtune. In the CLI, I wanted to decrease the value "gtune_loP_rll" below the standard of 10, because my copter tuned after a first run with Gtune already to this lower limit, and normally I also expect to have a lower roll P than 10 (1.0 in the GUI).

So I entered CLI and when I tryed to change this value to 5 I get an error:

# set gtune_loP_rll = 5
Invalid value

This happens also with gtune_loP for pitch and yaw axis. Everything higher/equal than 10 is accepted by the CLI. The used configurator is 1.1.0.
Is 10 the lowest value? The documentations says the range is [0..200].

Pierre-A, 2015-12-27:
**@droidparanoid**
see here    https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c
the min is 10 and max is 200.

droidparanoid, 2015-12-27:
Hi Pierre,

ok that explains the CLI response. But how can I use Gtune with values below 10? And the docs state:
"If you want tighter or more loose ranges change them here. gtune_loP_XXX can be configured lower than "10" that means a P of "1.0" in the GUI. So you can have "Zero P" but you may get sluggish initial control."

Was the range 0-10 originally existing and has been removed now?

closed 2017-02-25 by hydra