cleanflight/cleanflight #1629: light_led.c unused

MauroMombelli, 2015-12-27:
light_led.c seems to only provide the function initLeds() that does not get called anywhere

MauroMombelli, 2015-12-27:
also: startup_stm32f10x_md_gcc.S  and startup_stm32f10x_hd_gcc.S seems to be almost identical, only reset vector change; BUT in CMSIS sample (they are basically light revision of what can be found in TrueSTUDIO) those two file has major differences.

Is this ok? thepnly HD is EUSTM32F103RC_SRC, what board is this?

MauroMombelli, 2015-12-27:
in makefile, EUSTM32F103RC PORT103R (line 121) execute the same identical code than the else branch (generic f10x board like NAZE etc..) witch basically translate the if at line 75 instead to board-based, to MCU based (F3 or F1, and hopefully soon the F4)

hydra, 2015-12-29:
Hi @MauroMombelli - the last two comments belong in a seperate issue.

With regards to the main issue, thanks for the report. I'll look into this.

MauroMombelli, 2015-12-29:
moved issue to #1636  and #1635

commit eb9a401e (2016-01-08): Removing unused code.  Probably left over from F3 support.  Closes
#1629.
closed 2016-01-08 by hydra