cleanflight/cleanflight #1632: Proposal to improve numerical integration of ITerm in PIDs

martinbudden, 2015-12-28:
Currently the Cleanflight PIDs uses the simple [rectangle method](https://en.wikipedia.org/wiki/Rectangle_method) for integration. This, while simple to calculate, is only a first-order integration method and is numerically unstable (that is errors tend to accumulate rather than cancel each other out).

I propose, for `pidLuxFloat` we use [Simpson's method](https://en.wikipedia.org/wiki/Simpson%27s_rule), this is a fourth-order method and is both more accurate and more stable numerically. This is equivalent to using the [Runge-Kutta 4](https://en.wikipedia.org/wiki/Runge–Kutta_methods) method - this is fourth order and much more numerically stable.

Additionally I propose, that for the calculation of `ITerm` we use double-precision calculations, that is we use `double` instead of `float` (note this is just for the `ITerm`, not the `PTerm` or the `DTerm`). This is because, in integration, rounding errors accumulate. Numerical integration is one area where double-precision arithmetic is justified (I would argue it is necessary).

For `pidMultiWiiRewrite` I propose we use the [trapezium method](https://en.wikipedia.org/wiki/Trapezoidal_rule), I think the additional complexity of the Simpson method is probably not justified for an integer-based PID.

Note borisbstyle has already taken my trapezium rule integration changes for `pidLuxFloat` and `pidMultiWiiRewrite` for betaflight.

Note also I have a working implemetation of this which passes test code (and test code shows integration errors are significantly smaller). (Indeed one of the motivations of writing the test code was to enable the testing of the Simpson's method for integration.)

If people think this is a good idea then I'll create a pull request.

digitalentity, 2015-12-28:
+1 for this improvement!

ledvinap, 2015-12-28:
This does not make any sense.
The I-term is mainly for steady state compensation. It's accuracy is not (that) important, it should only drive error to zero. Try changing I term by few % - there should be negligible influence on flight performance.

Also data fed into PID are noisy - numerical 'noice' is negligible compared to physical one.

BTW: How do you want to compute error value between gyro samples? Input is data stream, not function ..

borisbstyle, 2015-12-28:
Here my tested changes.
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/flight/pid.c#L184
and PID1
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/flight/pid.c#L324

Personally I didn't notice any differences in this yet, but the idea kind of makes sense to me to make accumulation a bit slower. 
I did notice in past that higher Iterm can work against you in some faster acro manouvres. This will make it more tolerant to the small quick changes.
Still limited in testing of this.

martinbudden, 2015-12-28:
@ledvinap my concern was not accuracy, it was more to do with stability. That is the tendency of small errors to accumulate rather than cancel each other out. Simpon's method is more stable numerically than the rectangle method, so my assumption is that it is also more resilient to noise. That is the ITerm would be less susceptible to accumulate noise and become large. Of course this assumption needs to be tested with real data.

On second thoughts though I agree that given the noisy data double-precision arithmetic is not needed.

And on your final question, to get the midpoint value I use Simpson's over an interval of 2*dT, so the midpoint is available as part of the data stream. Of course this means that Simpson's method can't be used on an odd interval, so a trapezium approximation is used here, which is then wound back on the next even interval. As I said, this is working and tested. For your interest, here's the code (I've included some commented code to make it obvious what is going on):

        // -----calculate I component.
    #if defined(PID_INTEGRATE_SIMPSONS_RULE)
        float ITermDelta;
        if (evenInterval[axis] == true) {
            evenInterval[axis] = false;
            // for the even interval, use the proper Simpson rule
            // and then subtract the trapezium used in the previous odd interval
            //ITermDelta = (1.0f / 3.0f) * (rateError + 4 * rateError1[axis] + rateError2[axis]) * dT;
            //ITermDelta -= 0.5f * (rateError1[axis] + rateError2[axis]) * dT;
            // above two lines combined into one calculation
            ITermDelta = (1.0f / 6.0f) * (2 * rateError + 5 * rateError1[axis] - rateError2[axis]) * dT;
        } else {
            evenInterval[axis] = true;
            // cannot use Simpson's rule in an odd interval, so temporarily just use the trapezium rule
            ITermDelta = 0.5f * (rateError + rateError1[axis]) * dT;
        }
    #elif defined(PID_INTEGRATE_TRAPEZOIDAL_RULE)
        const float ITermDelta = 0.5f * (rateError + rateError1[axis]) * dT;
    #else
        const float ITermDelta = rateError * dT;
    #endif
        errorGyroIf[axis] = constrainf(errorGyroIf[axis] + ITermDelta * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);
[The #ifdefs are there so I can test all three methods.]

I accept this may be overkill and if that is the general opinion I'm happy to go with that - it was fun programming it anyway. If Simpson's method isn't used, I do think we should use the trapezium method - it's has little overhead and could provide some benefit.

ledvinap, 2015-12-28:
@martinbudden : There is negative feedback loop around PID (the controlled system closes the loop), so the error does not accumulate.   IMO there is no gain from improved accuracy.
Integrator is low-pass filter, it is very good at suppressing noise, numerical or physical ...

flounderscore, 2015-12-28:
@ledvinap : I totally agree with you. Higher order expansion of the integration is a waste of CPU cycles. First of all, in closed-loop the error integral should be close to zero anyway. But more important, the stability of the closed-loop system is not sensitive to small variations of the response of the loop filter (caused by the discrete time steps).

martinbudden, 2015-12-28:
I've written some test code some random noise is added to the angleRate and have integrated this over 10000 cycles. I can confirm that the ITerm does not accumulate error. There is random drift, but this is about the same whether Simpson, trapezoidal or rectangular integration is used.

Thanks for everyone's feedback. I'll close this issue.

borisbstyle, 2015-12-28:
Small error is not something to worry about for accumulation.

Can you run just one more test.

Try to make huge from positive to negative error jumps like when somebody
would quickly go inverted and back or fast acro rolls and flips.

I did see in the past where someone would have pretty high Iterm values
which would be fine in normal flight, but might cause overshoot in
agressive manouvre as Iterm would change too fast.

Openpilot crew already had Acro+ feature in their latest firmwares where
Iterm would get disabled above certain stick/throttle reflections to make
it better and more natural.

It's the Iterm giving the more "stiff" feeling in flight.

martinbudden, 2015-12-28:
OK, I'll reopen the issue while I do that test.

It will have to wait until tomorrow though...

borisbstyle, 2015-12-28:
I have no idea how much this will help, but just an idea. should make it
more steady.

I will try to find some bb logs showing what I exactly mean with this case.

ledvinap, 2015-12-28:
@borisbstyle: I posted link to article about handling I-term saturation some time back .. 

Maybe it would be interesting to back-calculate commands that are applied [after mixer] (use inverted mixer matrix). Then handle I-term saturation/speed up I-term change when user command is fighting I-term.
Maybe it will fix problem you describe  (and is mathematically quite clean).

digitalentity, 2015-12-28:
@ledvinap @borisbstyle 
Have a look at the PID controller I use for navigation https://github.com/digitalentity/cleanflight/blob/inav-dev/src/main/flight/navigation_rewrite.c#L1150

If we can use inverted mixer matrix to calculate constraints for PID outputs, I-term back-tracking will prevent windup most effectively.

martinbudden, 2015-12-29:
@borisbstyle bb logs showing what you mean will be useful. I'll hold off writing the test until I recieve them - that way my test can be based on real data.

martinbudden, 2016-01-11:
@borisbstyle any update on the availability of those logs? If you don't have time to provide them them let me know and I'll close this issue.

martinbudden, 2016-01-12:
To close this issue I'll create a pull request to add trapezium integration. This is cheap, has already been implemented in betaflight and may provide some small benefit at the margins. It seems that it is not worth doing Simpson integration since it is more expensive and provides no additional benefit.

borisbstyle, 2016-01-12:
@martinbudden 
I got plenty of logs where Iterm can mess up dynamic flight behaviour like acrobatic flights.
I found a solution for that which I am still testing. I simply scale Iterm down with more stick input and starting from above 70% stick reflection. This completaly solves the problem with bumps after some flips and rolls and makes possible to use higher I gains.

ledvinap, 2016-01-12:
@martinbudden : Simple sum as PID integrator is adequate, there is no need to complicate that.

martinbudden, 2016-01-12:
OK, I'll just close this then.

closed 2016-01-12 by martinbudden