cleanflight/cleanflight #1633: Failsafe - FAILSAFE box mode does not actually trigger failsafe

digitalentity, 2015-12-29:
I was looking at this part of code: https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/failsafe.c#L187-L207

It occured to me that FAILSAFE mode won't actually trigger failsafe procedure. To be able to safely and reliable test failsafe procedure this must be fixed.

@ProDrone you probably know ins and outs of the code better than anyone here. Can you assist in fixing this?

digitalentity, 2015-12-29:
@ProDrone 
See 90e9297 for a possible fix.

ProDrone, 2015-12-29:
@digitalentity Triggering is indirectly done from RX here: https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/rx.c#L526

I did that on purpose to test/simulate a loss of signal. What do you think?

digitalentity, 2015-12-30:
Ah, I see. Perfect! Means that we still can signal a failafe to the FC even if our receiver continues to send valid signals on link loss.

closed 2015-12-30 by digitalentity