cleanflight/cleanflight #1634: CC3D No softserial

Butzi250nighrhawk, 2015-12-29:
I flashed CF v 1.9 on my CC3D over OP GCS. Everything fine, motors and esc work. But in Ports tab there are only USBVCP UART1 and UART3, no Softserial. When i choose a RX_Serial or RX_PARALLEL_PWM and reboot no rx mode is ticked. RX_PPM and RX_MSP stay when rebooting, but no softserial shows up. Before I flashed my FC with OP, I once did the Wizard of OP and the rx worked.
I dont have a usb to uart adapter and i´m not able to go back to OP firmware. When i hit rescue it detects the board, when it times out when i click upgrade or upgrade&erase. I tried to hit upgrade and upgrade&erase and then plug in the cc3d, then it loads the firmware on it but it fails to reboot (and erase) so GCS OP doesnt work, but im able to flash CF back so CF configurator can connect to FC. Any fixes? any chances to get back to OP or "fix" the rx and softserial port? If only one of the two Problems is solved, I am happy. I know its kinda a issue mixed with support, but pls help me.

Flugsicherung, 2015-12-30:
You must use Cleanflight Configurator for the settings. On the Configuration Tab you can switch on the Soft Serials and make the settings for your Reciever. OP GCS is only for flashing the Firmware to the CC3D Board.

SteveAmor, 2015-12-31:
@Butzi250nighrhawk if this has answered your support question, can you please close this issue.

hydra, 2016-01-08:
Please use forums for support.

closed 2016-01-08 by hydra