cleanflight/cleanflight #1636: linker script wrong?

MauroMombelli, 2015-12-29:
startup_stm32f10x_md_gcc.S and startup_stm32f10x_hd_gcc.S seems to be almost identical, only reset vector change; BUT in CMSIS sample (they are basically light revision of what can be found in TrueSTUDIO) those two file has major differences.

Is this ok? the only HD is EUSTM32F103RC_SRC, what board is this? someone that has it can test it?

hydra, 2016-01-08:
The PORT103R is also HD.

The EUSTM32F103RC is one of these:

http://world.tmall.com/item/44483968260.htm?spm=a312a.7700714.0.0.ubO9C8

hydra, 2016-01-08:
`diff -bua lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/startup/TrueSTUDIO/startup_stm32f10x_hd.s lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/startup/TrueSTUDIO/startup_stm32f10x_md.s`

```
--- lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/startup/TrueSTUDIO/startup_stm32f10x_hd.s	2015-11-14 01:07:16.380708800 +0100
+++ lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/startup/TrueSTUDIO/startup_stm32f10x_md.s	2015-11-14 01:07:16.382708900 +0100
@@ -1,17 +1,15 @@
 /**
   ******************************************************************************
-  * @file      startup_stm32f10x_hd.s
+  * @file      startup_stm32f10x_md.s
   * @author    MCD Application Team
   * @version   V3.5.0
   * @date      11-March-2011
-  * @brief     STM32F10x High Density Devices vector table for Atollic toolchain.
+  * @brief     STM32F10x Medium Density Devices vector table for Atollic toolchain.
   *            This module performs:
   *                - Set the initial SP
   *                - Set the initial PC == Reset_Handler,
-  *                - Set the vector table entries with the exceptions ISR address,
+  *                - Set the vector table entries with the exceptions ISR address
   *                - Configure the clock system  
-  *                - Configure external SRAM mounted on STM3210E-EVAL board
-  *                  to be used as data memory (optional, to be enabled by user)
   *                - Branches to main in the C library (which eventually
   *                  calls main()).
   *            After Reset the Cortex-M3 processor is in Thread mode,
@@ -50,7 +48,7 @@
 /* end address for the .bss section. defined in linker script */
 .word	_ebss
 
-.equ  BootRAM,        0xF1E0F85F
+.equ  BootRAM, 0xF108F85F
 /**
  * @brief  This is the code that gets called when the processor first
  *          starts execution following a reset event. Only the absolutely
@@ -187,23 +185,6 @@
 	.word	EXTI15_10_IRQHandler
 	.word	RTCAlarm_IRQHandler
 	.word	USBWakeUp_IRQHandler
-	.word	TIM8_BRK_IRQHandler
-	.word	TIM8_UP_IRQHandler
-	.word	TIM8_TRG_COM_IRQHandler
-	.word	TIM8_CC_IRQHandler
-	.word	ADC3_IRQHandler
-	.word	FSMC_IRQHandler
-	.word	SDIO_IRQHandler
-	.word	TIM5_IRQHandler
-	.word	SPI3_IRQHandler
-	.word	UART4_IRQHandler
-	.word	UART5_IRQHandler
-	.word	TIM6_IRQHandler
-	.word	TIM7_IRQHandler
-	.word	DMA2_Channel1_IRQHandler
-	.word	DMA2_Channel2_IRQHandler
-	.word	DMA2_Channel3_IRQHandler
-	.word	DMA2_Channel4_5_IRQHandler
 	.word	0
 	.word	0
 	.word	0
@@ -211,45 +192,8 @@
 	.word	0
 	.word	0
 	.word	0
-	.word	0
-	.word	0
-	.word	0
-	.word	0
-	.word	0
-	.word	0
-	.word	0
[... 242 lines cut ...]
-	.thumb_set UART4_IRQHandler,Default_Handler
-
-	.weak  UART5_IRQHandler
-	.thumb_set UART5_IRQHandler,Default_Handler
-
-	.weak  TIM6_IRQHandler
-	.thumb_set TIM6_IRQHandler,Default_Handler
-
-	.weak  TIM7_IRQHandler
-	.thumb_set TIM7_IRQHandler,Default_Handler
-
-	.weak  DMA2_Channel1_IRQHandler
-	.thumb_set DMA2_Channel1_IRQHandler,Default_Handler
-
-	.weak  DMA2_Channel2_IRQHandler
-	.thumb_set DMA2_Channel2_IRQHandler,Default_Handler
-
-	.weak  DMA2_Channel3_IRQHandler
-	.thumb_set DMA2_Channel3_IRQHandler,Default_Handler
-
-	.weak  DMA2_Channel4_5_IRQHandler
-	.thumb_set DMA2_Channel4_5_IRQHandler,Default_Handler
-
 /******************* (C) COPYRIGHT 2011 STMicroelectronics *****END OF FILE****/
```

hydra, 2016-01-08:
What diff are you referring to? The above looks correct to me and certainly works on both the PORT103R and EUSTM32F103RC boards I have here.

MauroMombelli, 2016-01-10:
diff -bua lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/startup/TrueSTUDIO/startup_stm32f10x_hd.s 
src/main/startup/startup_stm32f10x_md_gcc.S

closed 2017-02-25 by hydra