cleanflight/cleanflight #1641: Cleanflight SPRacing F3 falls out of the sky

NicolasLaureyns, 2015-12-30:
Hello, 

I'm having a problem with my quad, on irregular base it just flips and falls out of the sky.. 
Sometimes when 5 min flying, sometimes 10 sec.. 

Turned on blackbox and here's an image  when it flips and falls: 
Could someone explain me what's happening ? 
[image]

Another one: 
[image]

You can find the whole blackbox logfile in attachment. 
[blackbox_log_2015-12-30_174732.TXT](https://github.com/cleanflight/cleanflight/files/75016/blackbox_log_2015-12-30_174732.TXT)
it's run 6 and 7. 

Small video when it flips: 
https://youtu.be/fyLdCBZ5yu4

Setup: 
Blackout 360 frame 
Cobra 2208 200Kv motors
SPRacing F3 FC (Running latest firmware 1.11.0) 
Kiss ESC 30A 
HQProp 6x4.5 Bullnose 
FrSky X8R receiver 
Turingy 3200Mah A-spec battery 

At first I thought it were the ESC's so I set the looptime from 2000 to 2500 to test, but still the same problem. 
Seems to me that the FC just stops at the end  on run 7? 

Thanks in advance.

joshuabardwell, 2015-12-30:
The log ends before the board disarms, which is why the traces end with square dots. This is a normal occurrence under flight conditions where there is too much data for the UART or the SD card to write at once. Most likely explanation is that the rapidly changing data that resulted from the copter flipping out overwhelmed the compression on the Blackbox link and overwhelmed the serial output. This isn't a bug.

Nothing in your Blackbox log seems to indicate a bug in Cleanflight. Most likely explanation IMO is a damaged gyro in the SP3 that is sporadically flipping out.

GitHub is for bug tracking, not support issues. If you agree that this is not a bug, please close this issue and get further support over on the RCGroups CleanFlight thread.

ledvinap, 2015-12-30:
@NicolasLaureyns : Where do you record blackbox data? Onboard chip or openlog?

NicolasLaureyns, 2015-12-30:
@ledvinap The onboard chip of the SP3. 
@joshuabardwell The board is brand new, the only crashes I made is because of the flipping..

ledvinap, 2015-12-30:
Maybe board resets? 
What state is the quad after crash?

How do you power the board?

NicolasLaureyns, 2015-12-30:
@ledvinap I reflashed the firmware on the board after some crashes, problem still presists. 
Gonna try to fly with the stock settings, tomorrow. 

The state of the board is disarmed, board is powered with Pololu 5V, 1A Step-Down Voltage Regulator D24V10F5

varoudis, 2015-12-30:
not sure if its related but I get reboot sequence with SPF3 when there is a lot a acc/gyro activity. Shaking the quad or slapping it a bit. All of this only when armed. If not armed (via switch) everything is ok.

There is a closed ticket on beta flight repo.

hydra, 2016-01-08:
It's probably a wiring issue or a defective board.  Which retailer did you buy the board from?  Have you contacted them?

closed 2016-01-08 by hydra