cleanflight/cleanflight #1647: Can't set PID values using CLI

paulwas, 2016-01-02:
Hi,
I downloaded and built the latest code so I could use the GTUNE fix for my Naze32. Then, when I was trying to set the PID values using the CLI, I found that I could only set them to 200. Looking at lines 619-627 in "serial_cli.c" I noticed that each value is set to "{ PID_MAX,  PID_MAX }". After changing them to "{ PID_MIN,  PID_MAX }" it seems to work fine. I looked for other obvious changes like that but didn't notice any. Anyway, I just wanted to pass that along.

Paul

SteveAmor, 2016-01-02:
Good catch @paulwas 

@NightHawk32 do you want to raise a PR for this ;)

paulwas, 2016-01-02:
I'd be happy to. However,. I'm new to this process so I'm now sure about exactly what I need to do. Could you give me either a few quick bullet-points or a pointer to the steps?

Paul

SteveAmor, 2016-01-03:
@paulwas you can follow the instructions here to create the PR to fix this issue:

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md#using-git-and-github

SteveAmor, 2016-01-04:
@paulwas - any update?  It's important this gets corrected in the code asap

paulwas, 2016-01-04:
Here's the pull request: https://github.com/cleanflight/cleanflight/pull/1655
Hopefully I did it correctly since I'm new to this. :-)

SteveAmor, 2016-01-04:
Looks good to me :)

NightHawk32, 2016-01-05:
Oops ;), thanks for the fix

paulwas, 2016-01-05:
Happy to help. :-)

hydra, 2016-01-08:
Good catch. PR merged.

closed 2016-01-08 by hydra