cleanflight/cleanflight #1651: MSP has no reserved values for expansion past one byte command

martinbudden, 2016-01-03:
The *Multiwii Serial Protocol* (MSP) is a **Length-Type-Value** protocol that uses one byte for the command type. This means it is limited to 255 commands. This means it is possible to run out of commands. Currenly 104 commands are `#defined` in `msp_serial.c`. Some of these are the Multiwii original commands, some are Baseflight commands and some are Cleanflight commands.

It may seem there are plenty of commands left, but the time to address a shortage is well before things start running out. There is a proposal to remove the command line interface (CLI), if this happens there will be more reliance on MSP and an increased demand for MSP commands.

The traditional way to allow expansion of a one byte type is to reserve values for expansion, namely:

254 - the following byte is a single byte value, offset by 256 - this gives and extra 256 commands
255 - the following two bytes are a two byte value - this gives an extra 65536 commands

I suggest we add the following to `serial_msp.c`:
```
#define MSP_EXTENSION_ONE_BYTE            254   // reserved for extension to allow extra byte for command
#define MSP_EXTENSION_TWO_BYTES           255   // reserved for extension to allow two byte commands
```

This is slightly complicated by the fact that 254 is currently used by `MSP_DEBUG`. This could be resovled in one of two ways: changing `MSP_EXTENSION_ONE_BYTE` to 252 (a currently unused value), or changing `MSP_DEBUG` to use a different value. I favour changing `MSP_DEBUG`, since I doubt this will effect ordinary users and it allows the traditional use of the values 254 and 255

digitalentity, 2016-01-03:
CLI removal will require 3 more MSP commands. However, I like the idea of having ability to expand beyond 256 commands.

martinbudden, 2016-01-11:
@hydra are you OK with changing the MSP_DEBUG value? If so I'll submit a PR

martinbudden, 2016-01-14:
I notice you've labled this as an enhancement. That suggests you may have misinterpreted my intent. I was not suggesting that we actually implement multi-byte commands, only that we reserve two command values to allow a future implementation. The easiest way to explain this is via code, so I've submitted a pull request. See #1682.

The thing I have reluctantly done is changed `MSP_DEBUG` from 254 to 252, to allow 254 to be used as one of the reserved commands. Strictly speaking this breaks compatibility, but it only does so for debugging so I think it is probably acceptable.

hydra, 2016-04-16:
i think we should just defer this until the first time we actually need it.  YAGNI springs to mind here.  Gonna close it, I'm not saying no, just not yet. :smile:

closed 2016-04-16 by hydra