cleanflight/cleanflight #1656: Naze32 CLI states invalid value

Sanguine7, 2016-01-05:
I followed Painless360'S vid on how to setup the naze 32 with frsrky telemetry but the cli only allows me to enable softserial and telemetry when i want to set the baud rate or do more the cli shatters every comand as "invalid value".
I hope this is just me having overseen something and someone can help

readerror67, 2016-01-05:
Maybe you should like, paste the input/output instead of referencing a video.

Sanguine7, 2016-01-05:
Sorry my bad,
D4R-II is in PPM mode connected to naze32, RX cable is soldered to pad 6
then i followed the guidelines in option 2
http://www.netraam.eu/nazewiki/pmwiki.php?n=Howto.FrskyTelemetry

feature SOFTSERIAL
feature TELEMETRY

those two where accepted

set softserial_baudrate=9600
set softserial_1_inverted=1
set telemetry_port=1
set telemetry_provider=0

for those 4 the CLI states "invalid value"

SteveAmor, 2016-01-05:
Just read the release notes

NOTE: CLI get/set commands have changed. some variables are now set using OFF/ON instead of 0/1.

This applies to telemetry.

Sanguine7, 2016-01-05:
Doesn't work either and this does not explain why i can't set baud rate
[image]

lostcontrol, 2016-01-05:
*softserial* does not exist (anymore I guess) as CLI command in Cleanflight. Look at https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md

SteveAmor, 2016-01-05:
The instructions you are following were last updated on July 02 2014. Lots has happened since then.  Those commands no longer exist. Have a look at:

https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md
https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md

Failing that, please ask support questions at RC Groups or IRC #cleanflight.

Sanguine7, 2016-01-05:
OK so i get that a lot has changed
@SteveAmor the comands given in your links don't work either
same problem that he does not know the command telem inversion given in link 2

So can anybody give me a link or commandline which enables me to use my frsky d4r-ii via softserial port 6? without external inverter?

i've seen that i can activate softserial and telemetry in the setup and set my softserial to frsky telemetry with baud 9600 in the config but my taranis still shows me no telem data

hydra, 2016-01-08:
Please use forums for support.

closed 2016-01-08 by hydra