cleanflight/cleanflight #1658: Use Spektrum satellite lost frame confer as a substitute for RSSI

digitalentity, 2016-01-07:
Spektrum sat receiver doesn't have an RSSI output, but it's reporting lost frame counter in the first byte of the packet. Maybe we can calculate how it changes over time to give user an indication of link quality?

flounderscore, 2016-01-07:
+1
If that worked, it would make cf very unique!

NightHawk32, 2016-01-07:
That would be a great thing. 
It won't be as accurate as a real RSSI, because the frame drop will increase pretty fast, I fear. But it is better than nothing.
If you're using a RX and a satellite in diversity mode there are also antenna fades reported which would occur before dropping frames. But when just jusing a satellite I think there is no better way than integrating the lost frames.

Tom-Roocke, 2016-01-08:
This would be a great feature.
While it wouldn't be an indication of true reception strength, it could be used to detect the edges of reception / dodgy areas.

flounderscore, 2016-01-08:
I did a little research and there seem to be mixed opinions about the first byte of each data packet that @digitalentity was referring to. It seems to be different between DSM and DSMX. And I bet it depends on the manufacturer, too (Spectrum vs Lemon vs Orange). Has anyone (of the CF contributors) every tried to read the dropped frames/antenna fades or what ever it is transmitting?

digitalentity, 2016-01-08:
There's a lot of controversial info about first two bytes of the frame.
Most info I could find indicates that the first byte is somehow related to
frame loss. Second byte is different depending on DSM/DSMX mode.
I have an original Spektrum satellite and a Lemon satellite, need to do
some testing and see what they output in the first bytes in different modes.

digitalentity, 2016-01-11:
I've made tests with two available Spektrum Sat receivers - original Spektrum and LemonRx.

First byte is indeed a lost frame counter. With both Spektrum and Lemon it starts counting at zero and increses each time a bad/corrupted frame is received.

However, this counter does not overflow. When it reaches a certain value (255 for Spek, 230 for Lemon) it no longer increses no mater how bad radio signal is. Second byte has nothing to do with this lost frame counter, it is a fixed value, different between Spek and Lemon (probably some flags).

If we use this as RSSI indicator, we will end up in the situation when RSSI is all green, but rx signal is totally unreliable.

flounderscore, 2016-01-11:
I'm wondering what exactly is reported in the first byte. Horizon Hobby gives three candidates:
> Antenna fades – represent the loss of a bit of information on that specific antenna. Typically it’s normal to have as many as 50 – 100 antenna fades on any one of the antennas during a flight. If any single antenna experiences over 500 fades in a single flight, the antenna should be repositioned in the aircraft to optimize the RF link.
> 
> Frame Loss – represents simultaneous antenna fades on all attached receivers. If the RF link is performing optimally, frame losses per flight should be less than 20.
> 
> Hold – a hold occurs when 45 contiguous (one right after the other) frame losses occur. This takes about one second. If a hold occurs during flight, it’s important to re-evaluate the system, moving the antennas to different locations and/or checking to be sure the transmitter and receivers are working correctly. 

One might have to use a separate system with Spektrum RX, Satellite, and telemetry and sniff the serial communication and compare what's transmitted via telemetry. Not quite sure how one could distinguish the fades from the satellite and main rx.

digitalentity, 2016-01-12:
The first byte is probably a "Frame loss" counter. It is definitely not Hold (increases too fast).

flounderscore, 2016-01-12:
How about something like this: (sorry for the pseudo code. That would be my best guess to get something rssi-like from the Spektrum sat)

    float loss = 0.0;
    float tau = 1e-3; // Some exponention decay rate. Should be something around 5s or so to make sure you can see it on the OSD. Numeric value depends on how often the code is going to be called. (11ms VS 22ms).
    
    counterPrevious = 0;
    
    while(1) // Somewhere where the RC information is read from the serial stream
    {
        if (counterPrevious > 200)
        {
            loss = 1000.0 // When the counter rails, just set RSSI to "you're in trouble".
        }
        int counterCurrent = getSpektrumFrameLoss();
        int delta = counterCurrent - counterPrevious;
        loss = tau * (loss + delta / 200.0 * 1000.0) // Scale RSSI from 0 to 1000.0. Apply exponential decay
        counterPrevious = counterCurrent;
    }

digitalentity, 2016-01-12:
@flounderscore 
Setting RSSI to "oh crap" value is the safest way of dealing with saturation.

@hydra 
What do you think? Should I post a PR for a "fake" RSSI?

error414, 2016-01-13:
we can call it "SFL" (spetrum frame loss), or something like that, instead of RSSI, and will be clear what it is.

flounderscore, 2016-01-13:
You can call it that in the documentation, but I think it should be treated as RSSI in the code to be compatible with OSDs etc.

CerealKiller1, 2016-01-18:
Hi, have you all had a look at this DORA thing on ebay and how they calculate an "RSSI value" from spectrum flightlog data? They post the weights they assign to each event somewhere in their docs (fades, holds and losses all account for bigger changes in the displayed RSSI)

http://www.ebay.com/itm/321221945100?_trksid=p2055119.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT

flounderscore, 2016-01-18:
Correct me if I'm wrong, but I thought DORA connects to the data port on a full-size Spektrum RX, not individual sats.

CerealKiller1, 2016-01-18:
No sir, you're correct. I was just mentioning it as those guy's have already figured out how to get a at least somewhat reliable reading out of the same data. Yes they get it from the data port of al full rx but it's the same data set you're all talking here. I just thought maybe you guys could use that one aspect they've already figured out  (i.e. the weight each event is assigned in the coming-up-with-an-rssi-number algorithm) instead of reinventing the wheel. 
I mean, they're algorithm is apparently good enough they sell out of those things every couple months at $25 a pop so it must work.

proggod, 2016-05-07:
Would you be willing to submit this PR to raceflight fork against the release branch?

neonbjb, 2016-10-03:
I've implemented something like this over in BF:
https://github.com/betaflight/betaflight/pull/1162

It works great, but has some limitations as is mentioned in the patch notes. The most interesting of them is that the satellite doesn't report the fade counter when it is bound to your TX via the CF command line - only when it is bound as a satellite RX.

closed 2017-02-25 by hydra