cleanflight/cleanflight #1659: Naze32 rev 5 afro flight Channel 1,2,3,4  not working after updating

markitos020, 2016-01-08:
I upgraded my board and now channels 1-4 dont work. no input shown in clean flight and channel 2 always shows 840 reading all other channels show 1500 mid point. All AUX channels work. If I flash back to and older version all channels except ch2 work. This problem started to happen after updating. Tried flashing ALL clean flight versions and channel 2 wont work...

SteveAmor, 2016-01-08:
Can you be very specific which version you first upgraded to.

markitos020, 2016-01-08:
I was running cleanflight 1.6.0. I upgraded to 1.11.1, in the update process it failed. I Had to jump boot loader pins and was able to install 1.11.1. Instantly all stick channels stopped working, only aux channels worked. If I flash 1.7.0 or below the only channel that wont work is ch2. I tried flashing back to baseflight and all channels worked. Since I dont like base flight I tried flashing cleanflight 1.11.1 again and the same problem occurs. I gave up and tried to run baseflight again untill a new fc would get here to try and see if thats the problem. Now I have the same problem on baseflight ch2 not working. I know its not tx or rx problems since I switched rx ch 1 and 2 and still there is no data on naze ch2 input. 

I would also like to comment that when I try running cleanflight 1.11.1 not only do channels 1-4 dont work. But throttle channel displays a mid point of 898 instead of 1500 like all other channels. Trying to set mid rc to 1520 (futaba) still leaves throttle channel at 898. Before updating I had done 3 flight with no crashes. This problem occured right after trying to update.

SteveAmor, 2016-01-08:
Sorry about your problem. It is really a support issue, not a bug. I suspect, failsafe is activating or a hardware fault. Please can you ask support questions at RC Groups or IRC #cleanflight and close this issue.

markitos020, 2016-01-08:
Ok thanks

closed 2016-01-08 by markitos020

markitos020, 2016-01-09:
Hi today i got the qav 210. I went ahead and started setting up my tx and
within a minute the led lights started to smoke. Im running the tattu 4s
lipo. Im guessing this led does not support 4s?

joshuabardwell, 2016-01-09:
This question is not even remotely on topic for this forum. Did you post it here by mistake?

vulture1, 2016-01-19:
I just hit the same issue where with CF 1.1.0 I can't get channels 1,2,3,4 to work. I have two rev 5 boards (6 DOF and 10 DOF), a full lemon receiver (PWM), CF is configured for PWM. AUX channels (1-4) work fine. I tried two different breakout cables. I know the receiver is working fine since I tested it with a attached directly to it. Thx

Val

sigrokBlack, 2016-02-11:
Same problem here!

sigrokBlack, 2016-02-11:
markitos020 How did you solve the problem above before you got the magic smoke?

markitos020, 2016-02-11:
Well in my case for some reason my problem was hardware. A loose solder pad, how did this happen right after update, I dont know maybe a coincidence. I noticed my problem was the board when I used a frsky sbus inverter and all channels worked. So I did a continuity test and found there was no continuity from channel 2 pin up to a tiny chip were all the reciever signals go to. I fixed the solder and all good.