cleanflight/cleanflight #1662: Add support for KISS ESC 24A Race Edition onewire telemetry

olelille, 2016-01-08:
The new 24A Kiss ESC has a one wire telemetry port with ESC temperature, voltage, current, consumption and eRPM data. This can be useful information for blackbox, telemetry and feedback to the controller. 

As i understand, multiple ESCs can be connected to the same serial port (UART) and the telemetry packets are triggered one by one with a 30µs pulse on PWM lead. 

More info can be found at:
http://www.rcgroups.com/forums/showatt.php?attachmentid=8524039&d=1450424877
http://www.rcgroups.com/forums/showthread.php?t=2555162

dkisselev, 2016-01-09:
These guys did a great job with this product, they made up a totally new feature that they know isn't supported by anything, but they wrote up a bunch of documentation on how to wire it up, how to handle their protocol in C, and even included an Arduino sketch.

Looks like you're right about a single UART and PWM-based triggering, the only relevant limitation might be that the telemetry is transmitted at 115200 baud, which isn't supported by softserial, but F3 targets make that easier with all their UARTs.

hydra, 2016-01-12:
@dkisselev yup. that's good work! they should be congratuated for providing all the details in a great way.

Definately having the information recorded in blackbox logs would be great to start with.  Having it displayed on an OSD would be icing on the cake.

pdelcast, 2016-02-01:
Castle has had this functionality for several years.  And releasing new quad code RSN.

http://www.castlecreations.com/support/documents/castle_link_live_2.0.pdf

closed 2017-02-25 by hydra