cleanflight/cleanflight #1665: Feature Request: Ring-buffer style Blackbox logging/Infinite recording for on-board SPI flash

flounderscore, 2016-01-09:
Would it be possible to implement some sort of a ring-buffer style Blackbox logging, where the beginning of the log is going to be overwritten when the memory is full? This would be a neat feature for people without external logger and to diagnose crashes etc.

What would be the best way to keep track of the current file position without writing to the same place on the flash over and over again?

ledvinap, 2016-01-09:
It is impossible to erase memory when recording blackbox - erase is too slow, there is not enough RAM to store blackbox during erase (there is already issue covering this).

It may be possible to erase flash when disarmed (for example using stick command), so you can discard unneeded logs.

End of recording is searched in current code.

flounderscore, 2016-01-09:
I was thinking to plainly overwrite the existing data. Imaging the following scenario:
* Format flash: fill everything with 0s (not really necessary) and set the address/file pointer to the beginning (0)
* Write data to flash and move pointer along.
* Once you hit the end of the flash memory, reset the pointer to the beginning and keep writing
* Now after a power cycle, the log should start were it left off previously (!= 0)

HaukeRa, 2016-01-09:
Overwriting is impossible in flash memory. If you overwrite flash it will result in a bitwise AND of the previously written data and the new data. A flash erase takes more time. During that time no read or write is possible for most flash devices.

flounderscore, 2016-01-09:
Thanks for clarifying.

Closed.

closed 2016-01-09 by flounderscore