cleanflight/cleanflight #1670: naze32 rev5 CF1.11 no response from the bootloader programming failed

ciscoe, 2016-01-11:
having issues with bootloader

trying to flash a r5 nase32 with cf1.11, not have any luck at all 

running cf 1.1.0

help please

joshuabardwell, 2016-01-11:
This is almost certainly a support issue, not a development issue.

https://github.com/cleanflight/cleanflight/blob/master/CONTRIBUTING.md

>Please remember the issue tracker on github is not for user support. Please also do not email developers directly for support. Instead please use IRC or the forums first, then if the problem is confirmed create an issue that details how to repeat the problem so it can be investigated.

ciscoe, 2016-01-11:
where do i post issues

JohnOCFII, 2016-01-12:
> where do i post issues

Your best bet is to use the RC Groups Cleanflight thread:  http://www.rcgroups.com/forums/showthread.php?t=2249574

closed 2016-01-12 by hydra