cleanflight/cleanflight #1675: Storage of PIDs as 8 bit values does not allow sufficient range or resolution tuning pidLuxFloat

martinbudden, 2016-01-12:
Currently the cleanflight configurator uses 8 bit values for the PID values (both in the UI and the MSP). This does not really provide sufficient range or resolution for PID tuning. There are already workarounds in the cleanflight code because of this (multiplying the ITerm by 10 #666) and boris finds that the D values are currently too aggressively scaled.

To improve PID tuning the configurator should use 16 bits for the PID values. Compatibility with previous versions needs to be maintained, so this suggests the addition of two new MSP commands, say `MSP_PID_FLOAT` and `MSP_SET_PID_FLOAT` to go alongside the current `MSP_PID` and `MSP_SET_PID` commands. There would also need to be corresponding changes to cleanflight configurator. See https://github.com/cleanflight/cleanflight/issues/1625 for previous discussion.

Since this would require a api level push, it makes sense to include this before the next release, because the api level has been pushed for current developments anyway.

There are several possibilities for increasing the range and resolution of the PID values:

1. ledvinap suggests using 16 bit floating point values
2. digitalentity suggests using binary fixed point values, Q6.16 format (which gives a range of -32 to 32 and a resolution of ~0.001)
3. use floating point values
4. use decimal fixed point values (that is, say, multiply the value by 100 before storing, divide by 100 on loading)

My thoughts:

1. 16 bit floating point values are not natively supported in either C or JavaScript. This means adding additional code to support them. This is not in itself a problem for JavaScript (ie the configurator), but will add code size to the firmware, which is undesirable. Additionally 16 bit floating point values give large increases in range at the expense of resolution, which is not really the trade-off we are looking for.
2. The problem with binary fixed point values is that they don't fully support round-tripping to decimal values. So a user could enter a PID of (say) 4.63 and then find the value is changed to 4.62 when they next look at it.
3. Floats probably require too much storage.
4. Decimal fixed point values seem to fit the bill. I suggest 2dp of resolution, this give a range of 0.0 to 655.35 for PIDs.
used

ledvinap, 2016-01-12:
half-precision floats (https://en.wikipedia.org/wiki/Half-precision_floating-point_format)
- conversion to integer is trivial, just mask mantisa / exponent and shift result, then apply sign. resulting value is exact from -2048 to 2048 and covers whole uint16_t and int16_t.
- conversion to float is even easier, just  move mantisa  and exponent into correct places of 32bit float.
- code size will be negligible, the functions can be shared in the code. The representation is only for transfer and storage, not for calculations.
- 11 bits of resolution(0-2048, plus sign) is a bit more than necessary to finetune most physical processes, resolution is larger that noise of sensors.
- resolution near zero is 1e-7 - this should easily cover parameters scaled to looptime (but may cause problem if scaling to microseconds is needed)
- range covers  uint16_t with reasonable resolution (32 per step) in upper end, so half float can replace uint16_t.
- special format for parameter storage may improve code quality - developers will be forced to use explicit store/fetch, forcing better isolation of code.

Full floats ar another possibility, but double storage space may be a bit of problem.

martinbudden, 2016-01-15:
@ledvinap to reply to your points:

* conversion to `int` and `float` may be trivial, but not as trivial as you suggest. The MSP is part of the public interface of Cleanflight, so a full implementation is required (ie including NaNs and subnormal numbers). Example code is here: https://stackoverflow.com/questions/1659440/32-bit-to-16-bit-floating-point-conversion. At a guess that looks like 68 bytes of constant data and about 200 bytes of code. That's just conversion to and from `float`, but conversion to `int` is not required (since 16-bit floats are just required to store and retrieve the PID values).
* whether or not that is negligable is open to debate. Nothing is negligable when you are on the margin and several of the boards are are on the margin (CJMCU, CC3D). OK, things will be better when the CLI is removed, but worse again if iNav is added.
* the implementation cost is not just born by Cleanflight. The Cleanflight configurator will also have to implement support in Javascript
* Third party tools will also have to implement it. This presents a (small) barrier to further development of those tools. But my view is that we should not impose additional burdens on the development of third party tools if that burden can be avoided
* I don't buy the argument about improving code quality. In my experience forcing developers to use something has little influence of code quality. Besides, once the PID stuff is implemented, who else will use this?

I actually quite like the idea of using 16 bit floats. But I think the downsides are significant and outweigh the benefits. So I don't think they are the best solution to this particular issue.

ledvinap, 2016-01-15:
This seems like reasonable implementation, it may be stripped down a bit:
https://github.com/numpy/numpy/blob/master/numpy/core/src/npymath/halffloat.c#L466
https://github.com/numpy/numpy/blob/master/numpy/core/src/npymath/halffloat.c#L235

As for javascript, quick search revealed this: https://gist.github.com/levskaya/1541883

Cleanflight currently avoid using floating point. Scaling is used, which is sometimes suboptimal. Having supported float format on both sides may help in lot of places ...

digitalentity, 2016-01-15:
Why not use fixed point values for config storage? Integer PID code would use the values as they are (only scaling will change to respect the decimal point). Converting fixed point to float point (where necessary) is a matter of doing a float point multiplication/division operation.

ledvinap, 2016-01-15:
@digitalentity : 
PID parameter dynamic range is over 10^3:
https://github.com/cleanflight/cleanflight/blob/bcd4800a0ccb4adb2e777b86332ee1941879eba6/src/main/io/serial_msp.c#L726-L743

An there are issues with range not being wide enough.

Also it would be nice to have I,D parameters in seconds, which would add some scaling to current values .. 

Other parameters may have similar problem - first developer used scaling to fit in 8bit interval, but it has little correspondence to physical units and range may not be enough for everyone ..

hydra, 2016-01-16:
I'm sitting on the fence right now.  I had a read up on half-floats, I'd not encountered them before so it was very interesting reading.  Amazing how much knowledge there is out there!

@ledvinap Can you perhaps point us at some areas (on both sides, firmware and gui) that would benefit from using half-floats.

As @martinbudden points out @eziosoft would also need to implement them in his (java) client.  Also the MinimOSD guys would need to do it too (arduino).  I have a feeling it would currently be received as painful by them.

martinbudden, 2016-01-17:
@hydra There are two separate issues here: 
1. What's the best solution for solving the PID storage problem (lack of range and resolution).
2. Would 16-bit floats be generally useful in Cleanflight?

It's important not to confuse the two.

To answer the PID storage problem we need to know what range and resolution we need/desire and choose a format that provides that. Additionally we wish to minimise any side effects of our choice of format.

16-bit floats are stored as a sign bit, a 5 bit exponent and a 10 bit matissa (giving 11 bits of significand precision). Storage of binary and decimal fixed point numbers is straightforward. Here is a comparision of their ranges and precision.

Representation | Max value | Min value | Binary precision | Decimal significant figures
---------------|-----------|-----------|-----------|----------------------------
[16-bit float](https://en.wikipedia.org/wiki/Half_precision_floating-point_format) | 65504 | 5.96 × 10^-8 | 11 bits | 3.31
[UQ6.10](https://en.wikipedia.org/wiki/Q_%28number_format%29) | 63.999 | 0.00098 | 16 bits | 4.82
[16-bit decimal fixed point](https://en.wikipedia.org/wiki/Fixed-point_arithmetic#Binary_vs._decimal), 2dp | 655.35 | 0.01 | 16 bits | 4.82
16-bit decimal fixed point, 3dp | 65.535 | 0.001 | 16 bits | 4.82
8-bit | 255 | 1 | 8 bits | 2.41

So the question remains: how much range and resolution do we need?

**Range**: The range is determined by the variation in PID values between different aircraft. (Note it is not determined by the variation in `P`, `I` and `D` values for a given aircraft since this can be addressed by scaling the differetn PID values separately. So, for example, if large `ITerms` and small `DTerms` are required that does not effect the storage requirements for PIDs generally.)

I have limited experience of tuning quadcopters, but on one occasion I put ArduPilot on a 250mm quadcopter and had to reduce the ITerm by almost a factor of 10 from its default (I guess ArduPilot is primarily used on larger quadcopters). If you take into account that Cleanflight runs on nanocopters all the way up to large aircraft it seems that it's not unreasonable to expect a dynamic range of at least 100. All the 16 bit representations provide sufficient range (all have a dynamic range of > 65000).

**Resolution**: The resolution is determined by how fine adjustment is needed when tuning an aircraft. All the 16-bit formats have more than 3 decimal significant figures of resolution. The 16-bit float trades some resolution for more range and I would say probably that resolution is more important than range, but probably not enough to be a deciding factor.

ShikOfTheRa, 2016-01-18:
Hi all, from my selfish perspective…. 

minimosd is native 8 bit so any float operations are avoided as will consume scarce resource resource. There is less than 1k of flash left and less than 100 bytes of SRAM.

Half float looks fantastic however if support that I may have to reduce functionality elsewhere.
Also will have to find or write something to deal with that. 

For me – if have to go change to accommodate, a standard 16 bit integer implementation would be easier ….

martinbudden, 2016-01-18:
@ShikOfTheRa, that seems to be a clinching argument against using 16-bit floats.

@hydra, I've done an implementation with as fixed point decimal, two decimal points. See: https://github.com/cleanflight/cleanflight/pull/1698

2 decimal points has the advantage that all values can be scaled by the same factor, making the code clear.

The code could be fairly easily changed to do 3 decimal points or QU6.10, but I think 2 decimal points is the best option. The code can serve as a basis for further discussion.

martinbudden, 2016-04-04:
@hydra , I now think we should drop this, in favour of changing `pidLuxFloat` to use the integer PIDs. 
as @borisbstyle has done in the latest version of betaflight, see [Luxfloat rework to int pids](https://github.com/borisbstyle/betaflight/commit/7b468c09f0d14978de4be201125ac4401bdb5776).

That is `pidLuxFloat` uses `P8[]` `I8[]` and `D8[]` appropriately scaled, rather than`P_f[]` `I_f[]` and `D_f[]`.

What do you think?

digitalentity, 2016-04-04:
I vote for this as well. Those two banks of PIDs are confusing, take config space and create a mess everywhere PID gains are used (MSP, adjustments, CLI)

borisbstyle, 2016-04-04:
Yes lets do that! In fact rewrite and luxfloat pids can get the same scale so the defaults work for both

ledvinap, 2016-04-04:
At least wrap `uint8_t` -> `float` conversion so that it is easy to replace/change ...

martinbudden, 2016-04-04:
@ledvinap , I would probably use defined constants rather than wraps, but agreed that I wouldn't just multiply by magic numbers.

@hydra , there seems to be a consensus that this is a good thing. What's your view?

If you like it, I'll implement it (with test code, of course).

I think this goes well with the parameter group work - this can be regarded as part of cleaning up the configuration settings for the PIDs.

martinbudden, 2016-04-06:
@borisbstyle , @digitalentity , @hydra, @ledvinap

I've made an implementation of this for discussion, see PR https://github.com/cleanflight/cleanflight/pull/2016

For test case, `PTerm` and `DTerm` calculated by `pidLuxfloat` and `pidMultiWiiRewrite` are within 1% of each other, `ITerm` within 2%. (Of course, more test cases are required, but the principle has been demonstrated.)

digitalentity, 2016-04-07:
@martinbudden thanks! Looks good to me.

hydra, 2016-04-07:
agreed.  i will close this pr.

closed 2016-04-07 by hydra