cleanflight/cleanflight #1679: little documentation problem

latomgo, 2016-01-14:
Hi,
the heading of the "Board - CC3D" section is wrong formatted in the documentation:
[image]
It's hard to see that a new section starts.
Thomas

SteveAmor, 2016-01-14:
Did you want to have a go at raising a pull request to fix? 

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md#using-git-and-github

latomgo, 2016-01-14:
this is just a hint, do whatever is necessary to correct the documentation.
I'm a user, no developer, no github knowledge at all. sorry, if this is the wrong place for my issue.

SteveAmor, 2016-01-14:
That's cool. It's a documentation issue that needs fixing. I'll take a look later :)

SteveAmor, 2016-01-14:
@latomgo PR raised.  Could you close this issue please (to keep things tidy).

latomgo, 2016-01-14:
ok, thank you

PR #1681 (2016-01-16): Documentation - a couple of manual.pdf fixes
Correctly added MotoLab board to pdf manual.

Fixed the heading issue for Board - CC3D due to line ending on Board - AlienWii32 (fixes #1679).
closed 2016-01-14 by latomgo