cleanflight/cleanflight #1680: Option to allow arming regardless of current attitude

iforce2d, 2016-01-14:
Unless I'm mistaken the default behavior is not to allow arming when the frame is not reasonably level. My quad is spending the night at the top of a very tall tree with a hard-to-climb lower section, and I'm 98.4% certain it could have thrashed its way loose if could just have armed it... but it was facing downward.

ledvinap, 2016-01-14:
just set 'small_angle' to 180 ..

But you may end up spending night at hospital ...

iforce2d, 2016-01-14:
aha... that's great, thankyou.

closed 2016-01-14 by iforce2d

joshuabardwell, 2016-01-14:
You can use stick commands to re-calibrate the accelerometer, then the copter will believe it's level and you can spin the motors.

iforce2d, 2016-01-15:
oh yeah, that's even better, wish I had realized that before the battery died. Oh well, I'm sure the next big storm will bring it down.