cleanflight/cleanflight #1684: custom tricopter for vtol

taileron, 2016-01-16:
till CF 1.11 the three flight situations (hover / transition / forward flight) can be partially and separate tested with the profiles three PID (rate profiles are not that important) settings are essential and probably three mmix and smix per profile. In CF 1.12 I couldn´t find pid profiles anymore. Currently the plane has to be landed between the flight situations ... an inflight mix/pid profile change would be great ...
If adding mixers per profile isn´t so easy, a three point aux switch per cli mixer line could solve it as well.
But then the limit of not exceeding 12 smix cli entries is insufficient.
customtricopter offers since 1.11 up to 4 really usable esc outs but only one of the two remaining servo outs works, the second shows only a fixed pulse with 1.5ms. 

so customtri could be used for a custom quad with one tilt servo

Is there a way to give custom_tri  3 esc and 3 servos outs ?
Otherwise it can only be done with more than one fc.

hydra, 2016-01-16:
Your request is *very* hard to understand except for the last part - `Is there a way to give custom_tri 3 esc and 3 servos outs ?`

Probably, is the answer.  However this might have an impact on other aspects of the sytem right now so not yet.  #884 needs to be completed first.

taileron, 2016-01-18:
sorry for my bad english and the way of stumbling in. A few years ago we tried to do a fc with components of that time but the complete source code got lost on the way …
https://www.youtube.com/watch?v=rRZFcrikoxg
Transition back often ended in a crash

Then I was searching for years till I tested old MWC with medium success but after the first tests with Naze and CF now it flies absolutely brilliant and rockstable at up to 35 degrees tilt (custom_ti with tilt forward servos controlled by aux channels) 

In tilt flight mode pitch is mixed to an aux in the tx and 3…5 % reverse pitch to the pitch channel to compensate the side effects of the tilt procedure to maintain the aerodynamic lift of the body

While doing a lot of research and testing one important effect was found: all kind of tilt mechanism components change the axis of thrust force and the axis of stabilization every time in the same way. So the same change of the mixers could be used for all applications.
Here is an example mmix of a quad with tilt mechanism.

Tilt full upward
mmix 0 1.000 -1.000  1.000 -1.000 (Rear R)
mmix 1 1.000 -1.000 -1.000  1.000 (Front R)
mmix 2 1.000  1.000  1.000  1.000 (Rear L)
mmix 3 1.000  1.000 -1.000 -1.000 (Front L)

no differential tilt is needed
smix 0 5 8 100 0 0 100 0  (Front Tilt Servo) (Tilt AUX channel)
smix 1 6 8 100 0 0 100 0  (Rear Tilt Servo) (Tilt AUX channel) 

Tilt full forward
mmix 0 1.000 -1.000 0.000  1.000 (Rear R)
mmix 1 1.000  1.000 0.000  1.000 (Front R)
mmix 2 1.000  1.000 0.000 -1.000 (Rear L)
mmix 3 1.000 -1.000 0.000 -1.000 (Front L)

smix 0 5 8 100 0 0 100 0  (Front Tilt Servo) (Tilt AUX channel)
smix 1 6 8 100 0 0 100 0  (Rear Tilt servo)   (Tilt AUX channel)
# mixer for differential tilt:
smix 2 5 1 100 0 0 100 0  (Front Tilt Servo) (Pitch Tilt down)
smix reverse 5 1 r
smix 3 6 1 100 0 0 100 0  (Rear Tilt Servo) (Pitch Tilt UP)

The common thing is: 
From full up to full forward pitch changes from differential thrust to differential tilt, Yaw changes to inverted Roll and Roll to Yaw. 

It would be absolutely great if an aux channel could control mixer rates to follow the tilt angles for having a proper control force in each tilt angle.

taileron, 2016-02-01:
probably it is possible to have 4 motors and two servos -  mmix 0 ...3 and two smix outputs then the vtol toy box would be a bit more comfortable. 
An inflight virtual tilt of the fc itself by controled an aux channel would help as well

closed 2017-02-25 by hydra