cleanflight/cleanflight #1685: Intermittent arming and oscillation on throttle up

bbbenji, 2016-01-16:
For some reason CF 1.11.0 as well as at least BF 2.3.0 and 2.3.1 are causing me issues with my Flip32+ v2.3. Arming is intermittent wether I set a switch or use the stock method with thumbstick. Once it actually does arm I get extreme oscillation when I give throttle, it stop when I relax on the throttle. This happens right from start with low values on throttle.

Screenshots: http://imgur.com/a/URLWR
Video: https://youtu.be/0Gb7qQht6x0
Dump: http://pastebin.com/TVAsNVHP (It's from BF but the same issue with CF)

Flashing CF 1.10.0 have resolved these issues for me.

SteveAmor, 2016-01-16:
@bbbenji can you perform a git bisect to find out exactly which commit caused this issue?

ledvinap, 2016-01-16:
@bbbenji : Are signals on RC tab stable and correct?

hydra, 2016-01-16:
That dump you provided was not a dump of the config you were trying to use.  I can see this from the 'aux' section - it's all defaults.

I suspect your receiver configuration is not correct and new cleanflight rx loss detection is either working perfectly.

Please reflash Cleanflight 1.12.0-RC1 via firmware flash (tick 'Show unstable releases') then configure it from scratch.
Please check that your receiver wiring is correct and that you have no faulty/intermittent connections.
Please provide a video showing the recevier tab.
Please provide a configuration dump (not log from configurator).
Please provide a photo of your wiring.

bbbenji, 2016-01-17:
@ledvinap RC signals are stable and correct. Low 1000 mid 1500 high 2000. Will bisect asap.
@hydra I edited the OP with a clean install dump with the same problem, that is why it is missing the other aux's. But the same problem exists. Intermittend arm and wobble on throttle up.
Wiring is correct. Proof: works on 1.10.0
Receiver tab is working as should.
Want to stress that it is working fine on 1.10.0 with no changes other than setting arm aux and ppm.

bbbenji, 2016-02-06:
I purchesed a Naze32 Rev6 clone. Hoping I would be able to get latest CF working. Unfortunately I am unable to! CF 1.11 and latest RC both cause extreme oscillation making it unflyable. It goes out of control.

I tried CF 1.10, as it worked on my Flip32, unfortuantely it does not recognize any of my sensors  on the Naze32.

I'm having trouble finding people with the same issue. Closest I could find is this: http://www.rcgroups.com/forums/showthread.php?t=2576384

bbbenji, 2016-02-06:
Video: https://youtu.be/ur65Dl_zALo
Pics: http://imgur.com/a/ErfW5

SteveAmor, 2016-02-06:
@bbbenji this is a support issue and should be raised in the support forums such as you provided a link for above. You will get much more support on RC Groups than here.  Please close this issue and reopen if you find a bug.

bbbenji, 2016-02-06:
@SteveAmor Considering the fact that on the Flip32, CF 1.10 works just fine compared to 1.11, I can't help but come to the conclusion that this is infact a bug. With no other variables being different.

bbbenji, 2016-02-07:
@SteveAmor Give me a few days to get a new order of props in and I will deliver the commit that started the problem. Thanks for brining git bisect into light for me. I have never heard of it before but it seems like a great tool.

bbbenji, 2016-02-08:
@SteveAmor After completing a git bisect the first bad commit is e347e7d4bf28ef15a214ea01643355910e39ce6f

Chadster, 2016-02-09:
I have similar results in both CF and BF.  I've never had good results with MW rewrite in any version, but after CF 1.10, and ~ BF 2.2.x (not sure of exact release), it's totally unusable and will result in a crash every time.  When I switch to Luxfloat, even with default PIDs, everything is smooth and locked in.

I stopped further testing cause I was going thru too many props.

bbbenji, 2016-02-09:
@Chadster Sorry you are having the same problem but I am glad I'm not the only one:) I have gone through a grip of props because of this problem myself. Which FC are you using?

Chadster, 2016-02-09:
Forgot to mention my setups - multiple NAZE32 FC's (acro and full), with Afro and ZTW ESCs. 

Luxfloat works good so I never really investigated it much, but didn't want you to think you were the only one with this issue.

Chadster, 2016-02-10:
This got me curious so I installed CF 1.12 RC2 on my Naze/ZMR250 and took a quick test spin.  Default settings on Luxfloat worked really great.  In fact my motors were running super smooth at 1000 loop time.

Landed and then switched over to MW rewrite and default settings.  Quad became immediately unstable upon launch and I killed the arm switch just to prevent more damage (was ready for this!).

I'm happy with Luxfloat so not too worried about all this.  However I do find it interesting that the default settings on those PIDs can have such wildly different results.

Chadster, 2016-02-10:
Tested a bit more and realized I had accidentally enabled angle mode instead of horizon, which I generally use as a default.  The combo of Angle and MW re-write PID resulted in a completely unstable quad.  Angle mode with LUX PID worked great.

I then tried Horizon mode with MW rewrite and found much more stable behavior and in fact it was quite flyable (though in need of PID tuning).  When I switched to rate mode stability improved further still.

Therefore it would seem that if you are using Angle mode you should probably not use MW PIDs.  This used to work fine before but I'm not sure at what version of Cleanflight it changed.  

BTW - with Betaflight, if I go back to 2.1.6 I can run stable with Angle + MW PIDs.  Anything beyond that version is unstable with that combo.  One more reason I guess to not use angle mode.

bbbenji, 2016-02-10:
@Chadster Thank you for further testing. When I have some free time in the future I may do another bisect to find when (rewrite) started causing the problems.

lostcontrol, 2016-02-11:
@Chadster What are your "P" setting for LEVEL? This is what is written in the documentation:

> In Angle mode, this controller uses the LEVEL "P" PID setting to decide how strong the auto-level correction should be. Note that the default value for P_Level is 90. This is more than likely too high of a value for most, and will cause the model to be very unstable in Angle Mode, and could result in loss of control. It is recommended to change this value to 20 before using PID Controller 1 in Angle Mode.

https://github.com/cleanflight/cleanflight/blob/master/docs/PID%20tuning.md

Have you tried to lower this value?

Chadster, 2016-02-11:
Ah, thank you for pointing this out!  I didn't notice that value in the
configurator, and I haven't reviewed the PID doc in a while.  It looks like
there have been quite a few improvements and additions.  Thanks again!

lostcontrol, 2016-02-11:
@Chadster Could you report if you test it? I plan to switch to MWREWRITE and this issue makes me a bit nervous :anguished:

lostcontrol, 2016-02-11:
FYI, I setup a new RotorX Atom this evening with 1.12.0-RC2, MWREWRITE and it works perfectly in angle mode. I lowered the "P" setting for LEVEL down to 2.0 instead of the default 9.0.

Chadster, 2016-02-14:
Tested MW-Rewrite again in Angle mode and this time lowered P-Level = 2.  This totally fixed the problem and my quad flew great.  Can now upgrade all my FC's without worry.

Thanks @lostcontrol for pointing out this value and not telling me I should have RTFM.

lostcontrol, 2016-02-14:
@Chadster I almost wrote it :smile: Glad to hear this solved your issue!!!

SteveAmor, 2016-02-14:
Is anyone going to raise a PR to change the default to 2?

bbbenji, 2016-02-14:
@SteveAmor I'll take care of it.

hydra, 2016-02-16:
pr merged.

documentation updated in https://github.com/cleanflight/cleanflight/commit/ba89d52268c971d4e17b7c943273f3c35dd7c714

hydra, 2016-02-16:
thanks everyone!

closed 2016-02-16 by hydra

IanGaming19, 2018-02-05:
I have tried to locate the "P" level in the PID tuning tab and cannot find it (Cleanflight). My quad flies fine in acro mode yet switching it into Angle mode causes lots of wobble. I build a 450 with 2212 dji motors and simonK 30a escs. I am also using a Naze32 rev5.