cleanflight/cleanflight #1690: Add telemetry output/voltage monitoring support for Radiolink receivers through I2C.

KP1533TM2, 2016-01-17:
Some Radiolink receivers, such as R9D and R10D, have an additional I2C port to plug in an addon module (PRM-01 or PRM-02).
* PRM-01 (http://www.radiolink.com.cn/doce/product-detail-111.html) only does battery voltage measurement;
* PRM-02 (http://www.radiolink.com.cn/doce/product-detail-110.html) requests telemetry data from the flight controller using PIX or APM protocol.

Receiver then sends that data back to the transmitter to display all that stuff on its screen.
There's no official documentation on I2C protocol those modules use, but few people managed to analyze PRM-01's communication and make a clone with Arduino (http://forum.rcdesign.ru/f4/thread416311-11.html).
Not much is known right now about PRM-02 module's data structures or communication protocol.

Technical details for battery voltage monitoring (from the same rcdesign.ru thread, post 406: http://forum.rcdesign.ru/f4/thread416311-11.html#post6127465):
* Receiver periodically reads from I2C addresses 0x01 to 0x05 (only when transmitter is on);
* PRM-01 module responds with 3 bytes on address 0x01 as follows: 1st byte is the cell count, 2nd byte is battery voltage high byte and 3rd byte is battery voltage low byte.
For example, response [0x03, 0x00, 0x73 (dec. 115)] means 3 cell battery measuring 11.5v.

It would be nice to have a feature to pass data (battery voltage, atleast) from flight controller to the receiver directly over I2C without having to use an addtitonal redundant module in the middle.

fsedarkalex, 2016-03-24:
+1 on this
RadioLink offers a chea and mighty alternative to FrSky's Taranis and it would be an overkill to also have tlemetry available for those!
I would be happy to test for you if you wanted me to do. Other idea which would be enough for me would be to just support talking to PRM-02 (PIX or APM-Protocol)

digitalentity, 2016-03-24:
@fsedarkalex talking to PRM-02 will be possible if #806 gets merged.

davidgrm, 2016-03-26:
I have manage to decode the messages between the PRM-02 module and the receiver. If this was implemented into cleanflight we would not need the module to decode Mavlink. We could plug the receiver's telemetry port directly into the FC I2C plug.
Would this be of any interest to anyone? I am just doing some final tests then I will release the data required.

KP1533TM2, 2016-03-26:
@davidgrm I am still interested in such implementation, however I'm not familiar enough with STM32's hardware and cleanflight source code enough yet to implement this myself.

davidgrm, 2016-03-26:
I know nothing about the workings of cleanflight but hopefully someone here does?? If I get a chance I will have a look and see if I can figure out where/ how one would add it. It would also depend on what else is using the I2C and if there is some way of turning it off

fsedarkalex, 2016-03-26:
unfortunately I don't know about CF-Internals eighther but I would also like to express my interest once again.

fsedarkalex, 2016-03-26:
For my FC (SP Racing F3 Deluxe) the manual says:

I2C1 connector - Used for external sensors and OLED displays.
The SCL and SDA are 3.3v signals.5.0v is always supplied via the on-board voltage regulators, even when powering via USB.
WARNING:  logic level converters are REQUIRED if your sensors require 5.0v signals.

So I think technically it should be possible to use it

davidgrm, 2016-03-26:
I did my testing with an AVR Xmega running on 3.3v and the receiver uses 5v. I connected the I2c pullup resistors to 3.3. This worked without problems.

KP1533TM2, 2016-05-09:
Just a little update.
Today I decided to see if I can implement battery monitoring the way I wanted (meaning connecting I2C bus of my Naze32 (V2.3_OEM) to R9D receiver I2C port). That failed (atleast at the moment).
Made an appropriate cable, hooked up a logic analyzer to debug stuff. Embarassingly enough, it took me an entire evening to figure out that all sensors are sharing this exact bus. R9D receiver acts as a bus master, and thus, can't be easily set up on the same I2C bus along with other master. In this case, an extra 'special' code that handles bus multimaster mode is required (that is, CPU would take over the bus to read sensors and stuff and then release it and act as a slave to respond when R9D asks for battery/telemetry info; that's how I see it working in my mind, but I'm not sure R9D would even check for bus activity before colliding with other data, since it seems it never was desingned to be used in multimaster mode).
So I looked into the possibility of using another I2C interface. That doesn't look very good either. According to STM32F103 datasheet, both I2C1 alternate and default pins are mapped to ports that are used to drive ESCs (again, on my Naze32 atleast) :(
The only obvious choice at this point seems (imho) to implement software interrupt-driven slave I2C, given that I'll find a free 'pin change' interrupt and enough corresponding free pins to physically plug the receiver's I2C into my Naze32.

rigo3t, 2016-06-01:
I'd be very interested in this too. I see that the X4RII receiver plugs in straight into the board and provides telemetry instantly, I wonder if this would ever be possible though. I'm not familiar with Cleanflight code or with RadioLink protocols, just stating my interest :)

FLYBYME, 2016-07-17:
@davidgrm Did you ever end up finishing the code? Im working on the same problem and wanted to see how you might have done it.

KP1533TM2, 2016-07-19:
I ended up using Radiolink PRM-02 module with MAVLink telemetry for now. It seeems to me, that the solution, for Naze32 atleast, would be quite non-trivial - it's either wasting 2 ESC ouputs to enable second I2C bus, or implementing I2C in software, or doing some weird multi-master stuff with sensor bus.

davidgrm, 2016-07-19:
@FLYBYME  I was not writing any code for it. All I did was figure out the mavlink commans that are being used so that is any of the coders here wanted to add the code they could

tjstyle, 2016-07-19:
@KP1533TM2 I more choose to make PRM-02 like module and talk using UART than directly using I2C.
I was try to make adaper code to translating MSP to PRM-02 format and flashed to Arduino Pro Mini, It's work fine.
And I think I will use i2c-gps-nav module with custom firmware to translating MSP data to PRM-02 data.

FLYBYME, 2016-07-19:
@tjstyle do you happen to have any code you care to share? I'm trying to do the same thing you are but I'm having trouble understanding the high and low bi bits. I don't understand the maths that must be taken from high and low bits to get a number the telemetry understands.

tjstyle, 2016-07-20:
@FLYBYME Still missing some value for the telemetry, I think I will share some code when all value was correct. VOLT, YAW,ROLL,PITC and ALT it was same like on CF Config. on  still don't know what value that I must send for SPEED, RISE and DIST. GPS is also correct
 including LATI and LONG, but for LATI the original value is with minus sign, I don't know why in the AT9 the minus sign is not shown.
[image]

FLYBYME, 2016-07-21:
@tjstyle  If you would mind just sharing the maths for the high and low bit for voltage that equals the 15.6v that you have showing in you images. I dont understand how  hi=44 lo=236 voltage=11.5v.

tjstyle, 2016-07-21:
@FLYBYME from where you've got hi=44 lo=236 voltage=11.5v ?

FLYBYME, 2016-07-21:
@tjstyle from this post; http://www.rcgroups.com/forums/showthread.php?t=2236519&page=14#post34359150

tjstyle, 2016-07-21:
I don't see any word about hi=44 lo=236 equal to voltage=11.5v
This is some example:
In multiwii serial protocol, the vbat value on MSP_ANALOG is UINT8, just convert that to UINT16.
When the the voltage is 11.5v the UINT8 value is 115, multiply that by 100 so you will have 11500 (UINT16) or 2CEC (HEX).
So you can use that with: hi=0x2C lo=0xEC

FLYBYME, 2016-07-21:
Oh I get it now. I didn't realize that hex have high and low bytes. My code is working now. I should have an arduino prm-02 working in the next few days.

tjstyle, 2016-07-21:
@FLYBYME great.. I'm glad to hear that.. Please let me know when you was made a working arduino prm-02 (clone)

FLYBYME, 2016-07-22:
@tjstyle I have the base code working. The only problem I have right now is that the mavlink library uses up about 80% of the memory of a arduino nano, and thats not including my code. Currently it only runs on an arduino mega. I have to find out a way to shrink the mavlink code so it does not take up so much space. Also I dont know how to send lat/lon data.

```arduino

#include <Wire.h>
#include <mavlink.h>

struct STREAM_DATA {
  uint16_t battVoltage = 0;
  float altitude = 0;
  float climb = 0;
  float roll = 0;
  float pitch = 0;
  float yaw = 0;
  uint8_t gps_sats = 0;
  float gps_lon = 0;
  float gps_lat = 0;
  float home_lon = 0;
  float home_lat = 0;
  uint16_t gps_speed = 0;
  boolean gps_fix = false;
  uint32_t home_distance = 0;
};

STREAM_DATA streamData;

boolean set = false;

void setup() {
  Serial.begin(57600);
  Serial2.begin(57600);

  Wire.begin(4);
  Wire.onRequest(onRequest);

  Serial.println("Starting up...");

}

void loop() {
  comm_receive();
}

void comm_receive() {

  mavlink_message_t msg;
  mavlink_status_t status;
  //receive data over serial
  while (Serial2.available() > 0) {

    if (mavlink_parse_char(MAVLINK_COMM_0, Serial2.read(), &msg, &status)) {
      // Handle message

      switch (msg.msgid) {
        case MAVLINK_MSG_ID_HEARTBEAT: {

            break;
          }
        case MAVLINK_MSG_ID_GPS_RAW_INT: {
            streamData.gps_sats = mavlink_msg_gps_raw_int_get_satellites_visible(&msg);
            streamData.gps_lon = mavlink_msg_gps_raw_int_get_lon(&msg) / 10000000.0000f;
            streamData.gps_lat = mavlink_msg_gps_raw_int_get_lat(&msg)  / 10000000.0000f;
            streamData.gps_speed = mavlink_msg_gps_raw_int_get_vel(&msg) / 100;
            /*
                        Serial.print("gps_sats: ");
                        Serial.println(streamData.gps_sats);
                        Serial.print("gps_lon: ");
                        Serial.println(streamData.gps_lon);
                        Serial.print("gps_latsats: ");
                        Serial.println(streamData.gps_lat);
                        Serial.print("gps_speed: ");
                        Serial.println(streamData.gps_speed);
            */
            if (!streamData.gps_fix) {
              if (mavlink_msg_gps_raw_int_get_fix_type(&msg) >= 3) {

                Serial.print("Setting home location: ");
                Serial.print(streamData.gps_lon);
                Serial.print("/");
                Serial.println(streamData.gps_lat);
                streamData.gps_fix = true;
                streamData.home_lon =  streamData.gps_lon;
                streamData.home_lat =  streamData.gps_lat;
              }
            }

            break;
          }

        case MAVLINK_MSG_ID_VFR_HUD: {
            streamData.altitude = mavlink_msg_vfr_hud_get_alt(&msg);
            streamData.climb = mavlink_msg_vfr_hud_get_climb(&msg);
            break;
          }

        case MAVLINK_MSG_ID_ATTITUDE: {
            streamData.roll =  mavlink_msg_attitude_get_roll(&msg);
            streamData.pitch =  mavlink_msg_attitude_get_pitch(&msg);
            streamData.yaw =  mavlink_msg_attitude_get_yaw(&msg);
            break;
          }

        case MAVLINK_MSG_ID_SYS_STATUS: {
            streamData.battVoltage = mavlink_msg_sys_status_get_voltage_battery(&msg);
            break;
          }
        case MAVLINK_MSG_ID_NAV_CONTROLLER_OUTPUT: {
            streamData.gps_fix = true;
            streamData.home_distance = mavlink_msg_nav_controller_output_get_wp_dist(&msg);
            break;
          }
      }

    }
    // And get the next one
  }
}
#define d2r (M_PI / 180.0)
float calc_dist(float lat1, float long1, float lat2, float long2) {
  double dlong = (long2 - long1) * d2r;
  double dlat = (lat2 - lat1) * d2r;
  double a = pow(sin(dlat / 2.0), 2) + cos(lat1 * d2r) * cos(lat2 * d2r) * pow(sin(dlong / 2.0), 2);
  double c = 2 * atan2(sqrt(a), sqrt(1 - a));
  float d = 6367 * c * 1000; // in meters
  if (d >= 0) {
    return d;
  } else {
    return -d;
  }
}

void set1() {

  int  alt = streamData.altitude * 10;
  byte  altHi = highByte(alt )  ;
  byte  altLo =  lowByte(alt ) ;

  int  yaw = streamData.yaw * 100;
  byte yawHi = highByte(yaw );
  byte yawLo = lowByte(yaw );

  int  speed2 = streamData.gps_speed * 10;
  byte speedHi = highByte(speed2);
  byte speedLo = lowByte(speed2);

  int  roll = streamData.roll * 10;
  byte rollHi = highByte(roll);
  byte rollLo = lowByte(roll);

  int  pitch = streamData.pitch * 10;
  byte pitchHi = highByte(pitch);
  byte pitchLo = lowByte(pitch);

  int distance = calc_dist(streamData.gps_lat, streamData.gps_lon, streamData.home_lat, streamData.home_lon) * 100;
  byte distanceHi = highByte(distance);
  byte distanceLo = lowByte(distance);

  byte buffer[16] = {0x89, 0xAB, streamData.gps_sats, altHi, altLo, yawHi, yawLo, speedHi, speedLo, rollHi , rollLo, pitchHi, pitchLo, distanceHi, distanceLo, 0x00};
  Wire.write(buffer, 16);
}
void set2() {

  int  rise = streamData.climb * 10;
  byte  riseHi = highByte(rise);
  byte  riseLo =  lowByte(rise);

  byte voltesHi = highByte(streamData.battVoltage);
  byte voltesLo = lowByte(streamData.battVoltage);

  float lat = 40.689060;
  float log = -74.044636;

  byte buffer[16] = {0x89, 0xCD, streamData.gps_sats, riseHi, riseLo, voltesHi, voltesLo,/**/ 0x0, 0x0, 0x0 , 0x0, /**/0x0, 0x0, 0x0, 0x0,/**/ 0x00};
  Wire.write(buffer, 16);
}

void onRequest() {
  if (set) {
    set = false;
    set1();
  } else {
    set = true;
    set2();
  }
}
```

tjstyle, 2016-07-22:
@FLYBYME Yes you need to remove unused mavlink data, It's better to you make your own repo for this code and we can talk in there, because this is out of topic from the issue above.

zarubik, 2016-08-05:
I put @FLYBYME's code to [MW OSD](https://github.com/ShikOfTheRa/scarab-osd) . My solution with Naze32 + MinimOSD Micro: https://github.com/zarubik/scarab-osd/commit/03f7e9d0b5e1ad1373917622ddc51f3342f8311b

Thx @FLYBYME 

[image]

tjstyle, 2016-08-05:
That's great.. @zarubik

FLYBYME, 2016-08-05:
@zarubik I see you got the lat/lon working I'm going to as that to my code.

clevertechru, 2016-08-11:
How i can get telemetry from my naze32 rev4 to r9d AT9 ?

omersiar, 2016-12-30:
@FLYBYME Can you share mavlink.h library file? or as whole to compile? How can we build your code?

FLYBYME, 2017-01-02:
I looked for my arduino folder and could not find it. Must have deleted it.

closed 2017-02-25 by hydra

pierrotm777, 2023-06-25:
> @FLYBYME Still missing some value for the telemetry, I think I will share some code when all value was correct. VOLT, YAW,ROLL,PITC and ALT it was same like on CF Config. on still don't know what [...]

Hello,

I try to build a telemetry for a friend that have an AT9 RadioLink and as you, i see that i have no minus character on the latitude.
Do you have found a solution ?
I use this code:
```
  union u32_tag  {
    uint8_t  b[4];
    int32_t ui32;
  } latit, longt;
  longt.ui32 = streamData.gps_lon;//V8
  latit.ui32 = streamData.gps_lat;//V8;
  uint8_t sat = streamData.gps_sats;// satelites count
  uint8_t bufferRlink[16] = {0x89, 0xCD,
                      sat, 
                      riseHi, riseLo, 
                      voltesHi, voltesLo,
                      longt.b[3], longt.b[2], longt.b[1] , longt.b[0],
                      latit.b[3], latit.b[2], latit.b[1] , latit.b[0], 
                      0x00};
```

Thanks for your help,

Pierre