cleanflight/cleanflight #1692: Scheduler optimisations

martinbudden, 2016-01-17:
I love the new scheduler. Brilliant!

I understand the people working on it know there are further optimisations possible, but wanted to get it out there so it can be used. I like the "get it working" and then "get it working well" approach: too much software is plagued by the problems of premature optimisation.

I normally like to keep issues atomic - that is only one item raised per issue. I've broken that rule here because I'd rather not fragment the discusion by having things discussed in multiple issues. This issue could serve as a thread for discussion of improvements to the scheduler.

I have not inconsiderable experience in the area of task scheduling, and especially cooperative task scheduling.

Here are some suggestions for improvement when time permits.

**Main improvement**: one common implementation for a task scheduler is to have a [run queue](https://en.wikipedia.org/wiki/Run_queue) of active tasks. Usually this is a [priority queue](https://en.wikipedia.org/wiki/Priority_queue), tasks with higher priority are nearer the front of the queue than those with lower priority. The Cleanflight scheduler would benefit from this approach: instead of having an `isEnabled` flag, tasked would be enabled or disabled by moving them in or out of the queue. Each time the scheduler was called it would only loop through all the enabled tasks, not all the tasks.

Implementing this is not a large piece of work, but neither is it trivial.

**In the small improvements** that could be implemented fairly easily now:

(1) Change the `cfTaskId_e` enum lists so that the tasks in the `cfTasks` array are in order of priority. This means that in the main scheduler loop the code that checks for realtime tasks:
```
/* Check for realtime tasks */
for (taskId = 0; taskId < TASK_COUNT; taskId++) {
    if (cfTasks[taskId].staticPriority == TASK_PRIORITY_REALTIME) {
        uint32_t nextExecuteAt = cfTasks[taskId].lastExecutedAt + cfTasks[taskId].desiredPeriod;
        ....
```
could change to something like:
```
/* Check for realtime tasks */
for (taskId = 0; taskId < TASK_COUNT; taskId++) {
    if (cfTasks[taskId].staticPriority < TASK_PRIORITY_REALTIME) {
        break;
    }
    uint32_t nextExecuteAt = cfTasks[taskId].lastExecutedAt + cfTasks[taskId].desiredPeriod;
     ....
```
ie, stop iterating through the cfTasks array when you get to the first non-realtime task.

(2) The lines:
```
bool outsideRealtimeGuardInterval = (timeToNextRealtimeTask > realtimeGuardInterval);
....
bool taskCanBeChosenForScheduling =
    (outsideRealtimeGuardInterval) ||
    (cfTasks[taskId].taskAgeCycles > 1) ||
    (cfTasks[taskId].staticPriority == TASK_PRIORITY_REALTIME);

if (taskCanBeChosenForScheduling && (cfTasks[taskId].dynamicPriority > selectedTaskDynPrio)) {
    selectedTaskDynPrio = cfTasks[taskId].dynamicPriority;
    selectedTaskId = taskId;
}
```
are more efficiently implemented:
```
if (cfTasks[taskId].dynamicPriority > selectedTaskDynPrio) {
    bool outsideRealtimeGuardInterval = (timeToNextRealtimeTask > realtimeGuardInterval);
    bool taskCanBeChosenForScheduling =
        (outsideRealtimeGuardInterval) ||
        (cfTasks[taskId].taskAgeCycles > 1) ||
        (cfTasks[taskId].staticPriority == TASK_PRIORITY_REALTIME);
    if (taskCanBeChosenForScheduling) {
        selectedTaskDynPrio = cfTasks[taskId].dynamicPriority;
        selectedTaskId = taskId;
    }
}
```

(3) The non-`NULL` test in
```
/* Execute task */
if (cfTasks[selectedTaskId].taskFunc != NULL) {
    cfTasks[selectedTaskId].taskFunc();
}
```
is unnecessary, just call:
```
cfTasks[selectedTaskId].taskFunc();
```
This is true because `taskFunc` is never null. And there is no cause ever to set it to `NULL`. And even if there was some reason to set it `NULL` then you could impose the rule that you also had to set `isEnabled` to `false`.

(4) In the scheduler loop in `main.c` there is a call to `processLoopback()`. Presumably this is historical and `processLoopback` should just become a task.

(5) Not an optimisation, but can we move the scheduler code into its own directory? Although it is small I think it is important enough to warrant this.

martinbudden, 2016-01-18:
For discussion, I've implented a priority queue for the scheduler, see:
https://github.com/cleanflight/cleanflight/compare/master...martinbudden:scheduler_queue

Note that this is completely untested and should not be used (there are also some kludges that need to be fixed). But if people like it I can clean it up, write the testcode and make a PR.

Note that this does not in any way change the underlying scheduling algorithm, it just keeps the tasks in a priority queue and so is a more effecient implementation. Efficiency is improved mainly for two reasons:

1. only enabled tasks are in the queue
2. the queue is in priority order

There are also one or two minor code optimisations.

henn1001, 2016-01-20:
@digitalentity is the right man for this :)

ledvinap, 2016-02-07:
IMO the priority concept as used in current code is a bit problematic. Available RTOS have strict priority ordering (higher priority task is always run before lower priority), scheduling on same priority level may be round-robin.

Cleanflight should adopt this 'limitation' early, so that porting is possible later.

If some 'backoff' strategy on high load is necessary, it may be implemented independently of main scheduler - maybe just increase time when task is scheduled again(but do not touch scheduled time).

If this change is adopted, it should be quite easy to take scheduller from FreeRTOS os ChibiOS and change it to cooperative mode. There will be tested and proven scheduler, it would avoid lot's of possible problems ...

closed 2016-04-07 by martinbudden