cleanflight/cleanflight #1694: Current meter isn't working properly in 1.12

int77, 2016-01-17:
Hi,

Battery Current meter has some problems since implementation of the new task scheduler.
Current is measured correctly (hardware sensor in my case), but capacity drawn is always shown as zero in both configurator and OSD. Didn't get time to dig into the code, but it looks like there is something wrong with increment of the static variable mAhdrawnRaw in function updateCurrentMeter.

Same hardware is working correctly with previous versions.

hydra, 2016-01-18:
thanks for the report, investigating.

@digitalentity @borisbstyle any ideas?

hydra, 2016-01-18:
It's working ok on stm32f3discovery.  @int77 what board are you using?

digitalentity, 2016-01-18:
@hydra 
the updateCurrentMeter() code didn't change, we only changed the way it is called. Current meter is working for me (CC3D) - both current and mAh drawn are updated as they should. 
@int77 
can you give some details? board and config dump would be great.

hydra, 2016-01-18:
Working for me here on Naze32 too.

int77, 2016-01-18:
I'm running SPRF3. That't what I'm getting in GUI:
[image]

Corresponding dump file:
[dump_18012016.txt](https://github.com/cleanflight/cleanflight/files/94994/dump_18012016.txt)

int77, 2016-01-20:
After some investigations, I’ve found the root cause. Actually, the issue is linked to my hardware.

The calibration of my current sensor in the middle / high current range requires high offset, resulting in slightly negative drawn current (variable amperage) measured in idle mode. In current implementation, variable amperage is constrained to non-negative values when sent over MSP, but is taken as is when used to calculate the increment of the drawn capacity (variable mAhdrawnRaw). 

It means mAhDrawnRaw can take negative amperage and be decreased in idle mode, leading to negative value of the drawn capacity mAhdrawn.  The mAhDrawn is also constrained to positive values when sent over MSP, and I see zero drawn capacity in GUI during some time, while actual value of the mAhDrawn is negative.

To avoid this kind of situation, the fix would be to constrain the output of the currentSensorToCentiamps() function to non-negative  values:
```c
int32_t currentSensorToCentiamps(uint16_t src)
{
    int32_t millivolts;

    millivolts = ((uint32_t)src * ADCVREF) / 4096;
    millivolts -= batteryConfig->currentMeterOffset;

    if (millivolts < 0) return 0;

    return (millivolts * 1000) / (int32_t)batteryConfig->currentMeterScale; // current in 0.01A steps
}

```  

Thanks for checks and feedback provided.

hydra, 2016-01-24:
can you please create a pull request for this.

ledvinap, 2016-01-24:
Truncating negative current values makes it hard to calibrate sensor properly. And is only masking wrong calibration.

It would be better to truncate negative values only when calculation battery capacity (amperage in updateCurrentMeter)

hydra, 2016-01-24:
Thanks for the ibfo @ledvinap I haven't had the time to properly digest everything yet.

int77, 2016-01-24:
The point regarding calibration makes sense.

However, the current is already constrained to non-negative values when sent over MSP with Multiwii Current Output option activated:
````c++
case MSP_ANALOG:
        headSerialReply(7);
        serialize8((uint8_t)constrain(vbat, 0, 255));
        serialize16((uint16_t)constrain(mAhDrawn, 0, 0xFFFF)); // milliamp hours drawn from battery
        serialize16(rssi);
        if(masterConfig.batteryConfig.multiwiiCurrentMeterOutput) {
            serialize16((uint16_t)constrain(amperage * 10, 0, 0xFFFF)); // send amperage in 0.001 A steps. Negative range is truncated to zero
        } else
            serialize16((int16_t)constrain(amperage, -0x8000, 0x7FFF)); // send amperage in 0.01 A steps, range is -320A to 320A
        break;
```
I don't know the history behind this imlementation, but it might be for compatibility reasons with some OSDs. Anyway, taking this into account, documentation could specify that current calibration has to be done with "mutliwii current meter output OFF", and it isn't the case.

Clean fix could then include two steps:

1/  Battery capacity calculations:
```c++
void updateCurrentMeter(int32_t lastUpdateAt, rxConfig_t *rxConfig, uint16_t deadband3d_throttle)
{
...
    //mAhdrawnRaw += (amperage * lastUpdateAt) / 1000;
    mAhdrawnRaw += (constrain(amperage, 0, 0xFFFF) * lastUpdateAt) / 1000;
    mAhDrawn = mAhdrawnRaw / (3600 * 100);
}
```
2/ Update documentation regarding current sensor calibration conditions with "mutliwii current meter output OFF".

ledvinap, 2016-01-25:
Does anyone use multiwiiCurrentMeterOutput? Documentation should mention its limitations..

ad 1) There is no need for 0xffff limit (655A should be enough for everyone), simply strip negative values
```
mAhdrawnRaw += (max(0, amperage) * lastUpdateAt) / 1000;
```

int77, 2016-01-26:
MinimOSD or clone boards running MW / Rush / KV team OSD or similar firmware could be affected by this issue.

I've created [7a074ff](https://github.com/int77/cleanflight/commit/7a074ff2ed34527002ee61af4603ff642668a16a) commit summarizing the discussion. Pls have a look and feedback if any further suggestion.

hydra, 2016-01-27:
Please create a PR with the commit.

int77, 2016-01-27:
[PR#1729](https://github.com/cleanflight/cleanflight/pull/1729) created.

closed 2016-01-27 by int77