cleanflight/cleanflight #1695: SONAR pins with SOFTSERIAL (conflict)

jelinj8, 2016-01-17:
Is there some objective reason to not have a FEATURE_SOFTSERIAL in sonar.c (about line 73):
    `if (feature(FEATURE_SOFTSERIAL ) || feature(FEATURE_RX_PARALLEL_PWM ) || (feature(FEATURE_CURRENT_METER) && batteryConfig->currentMeterType == CURRENT_SENSOR_ADC) )` ?
It seems to work OK on my FLIP32/NAZE. Without it the SONAR and SOFTSERIAL features are in undetected conflict, with the modification the softserial works together with sonar (pins moved to PWM56)

dkisselev, 2016-01-17:
I don't know the specifics in this case, but generally softserial is incompatible with other features because of a lack of hardware timers.

F1 boards only have 2(?) hardware timers which can be configured to trigger interrupts at precise intervals. I believe the flight control/rc/PID system uses one of the timers to fly, so the second one is free for additional features.

Softserial requires the second timer so that it can bitbang an accurate baud rate, and I think we're just missing the code to allow that timer to be shared with other features. I'm not sure about Sonar, but I know LED is one that also requires a timer to be able to drive the LED's at the precise right speed, so even though the RC pins might not overlap, if the two features are enabled at once they'll try to compete for when the timer is set to trigger and what action it takes, causing everything to just kinda fall apart.

ledvinap, 2016-01-17:
@dkisselev : F1 has 4 times each with 4 channels. But timer channels are used for PWM generation/receiving/ppm.
General timing is based on SYSTICK timer.

softserial timer is partially shareable, but I am not sure about details.
Sonar measurement is based on EXTI and systick, no TIM resources are needed.

RGB led driver needs dedicated timer (TIM) now, this timer can't be shared.

hydra, 2016-01-18:
Yes, this would be fine, there is code in PWM Mapping to exclude the sonar pins that are selected by the configuration you mention.  I will add this into 1.12.0 now.

hydra, 2016-01-18:
I also now make it select PWM5/6 when RSSI_ADC is used.

hydra, 2016-01-18:
fixed in 7aae53d8ebb6df65caad6d6a3122a8a1c19dbadd and 4ac96aeca416d16e8b2ab2299f75b79d311080b4

Please grab a new binary and test.

hydra, 2016-01-18:
This is another issue that really highlights the need for #884.

hydra, 2016-01-19:
@jelinj8 have you tested this yet, it's waiting your feedback.

jelinj8, 2016-01-19:
Sorry for delay, I've reflashed yesterday, it is working OK (I'm using it recompiled with this mod already since summer)

hydra, 2016-02-02:
great, thanks for feedback. will close this issue now.

closed 2016-02-02 by hydra