cleanflight/cleanflight #1699: with mixer CUSTOM servos not working at all

universam1, 2016-01-18:
I have a Tricopter with 2 servos for tilt, unfortunately CUSTOMTRI is not applicable since I can configure 2nd servo but the output is not activated.
So I want to choose CUSTOM mixer, but there even though I configure cli servo and smix no servo output is on any port. I'm using a Naze32 board.

When looking through the code in mixer.c on line 455 you can see that loadCustomServoMixer() is not called for MIXER_CUSTOM. The same problem I find on line 832 where no servoMixer() is called for case MIXER_CUSTOM.

According to the documentation all custom mixer should have SMIX enabled. Is this true or not? Any hint how to solve that?

hydra, 2016-01-19:
what board?

universam1, 2016-01-19:
Naze32 F1 acro

badfrogg, 2016-03-11:
I have verified this in Naze and SPRF3 while experimenting this airplane builds that have glow engines with servos for throttle. mixer "custom" has no smix.

I would like to see mixer "Custom" to be flexible for motors or servos in any combination including all servos or all motors.

hydra, 2016-03-12:
Use CUSTOM AIRPLANE mixer.

DzikuVx, 2016-05-13:
@hydra in iNav it was solved in inavflight#148

closed 2017-02-25 by hydra