cleanflight/cleanflight #1707: an issue about CC3D's UART3

xiaoqixia, 2016-01-21:
I flashed V1.12 fireware on my CC3D board last day and wanted to enable the UART3 to use GPS,but actually I find I can not enable the UART3.It will disable when the system saved and restart.when I enable the GPS and set baud in 115200 in Ports  or enable the GPS in Configuration,it happened again.I am not familiar with cleanflight but I am trying my best to learn it.So is there any wrong with me?why the flex port can not be enabled?

xiaoqixia, 2016-01-21:
PS:
a small question:in the Modes,why can't I find GTUNE ?

Flugsicherung, 2016-01-26:
Hi xiaoqixia,
I flashed V1.12.0 RC1 on my CC3D Board and reset settings to default.
On the Ports Tab I set UART3 on GPS to 115200 and it works. After Reset the Configuration is working too.You must flash the CC3D on Mainport with a FTDI Board.....

hydra, 2016-01-27:
Please use forums for support.

closed 2016-01-27 by hydra