cleanflight/cleanflight #1711: Failsafe - Voltage-activated failsafe

digitalentity, 2016-01-22:
I have received a few requests for adding RTH functionality on a "low battery" event to iNav firmware.
This issue is opened here and not in iNav as it this it would be better to implement a more generic "failsafe on a low battery event".
@ProDrone, @hydra what do you think?

lostcontrol, 2016-01-22:
As you probably know, Ardupilot has "do nothing", "land" and "RTH" options on battery fail-safe. I think you can now even define additional safe landing points and Ardupilot will chose the nearest one if it can't reach home again.

digitalentity, 2016-01-22:
@lostcontrol I know what APM is capable of, but this is about Cleanflight that does nothing except beeping on low battery.

lostcontrol, 2016-01-22:
@digitalentity Just wanted to list the available options there since Cleanflight will probably end up with similar behaviors. Also I always wondered why Cleanflight has *vbat_min_cell_voltage* and *vbat_warning_cell_voltage* since both seem to result in the exact same action (beeping)!? Or is it not possible to arm anymore if *vbat_warning_cell_voltage* is triggered?

DL6ER, 2016-01-24:
> Or is it not possible to arm anymore if vbat_warning_cell_voltage is triggered?

I looked into the source code and (currently) that is not the case. The beeper beeps differently for warning and critical levels. That's everything that seems to happen.

ledvinap, 2016-01-24:
Battery voltage (or remaining battery capacity) +  warning/critical level may be exposed as virtual AUX channel .. Then it will be trivial to configure it for desired action

closed 2017-02-25 by hydra

oleost, 2017-03-10:
@hydra Is this somehow solved since it's closed ?

Dutchy-79, 2017-06-02:
I'd like the idea of a low voltage failsafe trigger, I'm a APM pilot and now building a new hex with the omnibus F3 and inav.

Dutchy-79, 2017-06-02:
Can't edit the post above, but if the future isn't there yet I already had a new idea for this. 

I'll try making a logical switch on the Taranis to trigger the failsafe aux on a set voltage.

mikeller, 2017-06-02:
@Dutchy-79: I got exactly that on my TX. It's a voice alert that tells me if my voltage drops too low, prompting me to land (unless it happens on a long punchout, when the computer between my ears uses logic to determine that this is just expected behaviour).

Dutchy-79, 2017-06-02:
@mikeller I'm planning on making it automatically with the logical switch. 

Tho didn't think about the long punch out. Maybe I need to set a time period as well. Say <3.8v for 5 seconds triggers a logical switch on aux(failsafe channel)

mikeller, 2017-06-02:
Up to you really. In my opinion, if you have telemetry, having the TX read out the voltage to you periodically if it's below a certain level is still best. You as the pilot know best if it's only caused by an old and sagging battery, or if it's not wise to do a failsafe landing right above a large puddle.

Dutchy-79, 2017-06-02:
No, I was thinking of a RTH failsafe.

mikeller, 2017-06-02:
Ah, I see. Advanced navigation features like RTH are more or less unsupported, and certainly untested in cleanflight / Betaflight. If that is what you are after, you should have a look at iNav: https://github.com/iNavFlight/inav

Dutchy-79, 2017-06-02:
Okay thanks, I'll ask there again. I ended up here because I searched Inav RTH low battery failsafe on Google and the first 2 posts talked about inav so I didn't even look at the master branch ;) 

My bad,