cleanflight/cleanflight #1712: servo and smix rates

universam1, 2016-01-22:
Could you please check the rates for servo and smix CLI?

Refering to 
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.h#L156
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.h#L177
the servo rates should be extenseble to +/-125 - unfortunately they are limited to +/-100.

This is especially an issue with smix cli where I want a higher rate on different mixer channel.

See
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c#L1438
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c#L1305

Thank you for correcting!

universam1, 2016-01-22:
please consider #1713

hydra, 2016-05-06:
The links are now out of date.

closed 2017-02-25 by hydra