cleanflight/cleanflight #1716: GPS Possible With Non-CPPM Receiver?

CradyCopter, 2016-01-23:
I'm using a FrSky VD5M micro RX (*non-CPPM*) with a NAZE32 rev 5 board and was wondering if there is an alternate method of connecting GPS.

Maybe I missed something but all the wiring diagrams I've seem require a CPPM RX in order to free up pins 3 and 4 of the RC inputs for the GPS connections.

Thanks!

kilrah, 2016-01-23:
CPPM has nothing to do with GPS, and a GPS would connect to the telemetry/smart port, not the servo pins... but given the VD5M doesn't even have telemetry you obviously can't use a GPS with it.

By the way this is not an OpenTX technical issue, please use forums for usage questions.

CradyCopter, 2016-01-23:
Uh... WTF???
a) GPS doesn't require a RX with telemetry and b) I mentioned CPPM because as I stated, all diagrams [that I have seen] indicate that a CPPM RX is required in order to free up pins 3 and 4.

dan557, 2016-01-23:
This is a support question not an issue with cleanflight. Please ask on the cleanflight thread

http://www.rcgroups.com/forums/showthread.php?t=2249574

skaman82, 2016-01-23:
You can use the UART1 pins (see manual), but have to disconnect the gps to use the usb port.

closed 2016-01-27 by hydra