cleanflight/cleanflight #1717: Suggestion: Different VBAT pin for SPRacing F3 derivatives

Arakon, 2016-01-24:
I noticed several FCs came out recently that use the SPRACINGF3 target. However, some of these have no vbat pin (i.e. fpvmodel X-Racer). It would be great if that pin could be reassigned to another (similar to CC3D, where VBAT is on RC-In5, using a basic voltage divider). I had attempted to change it myself via the target.h, but apparently more work is needed, as the compiled code continues to monitor vbat on PA4 regardless of setting in target.h.
Either Motor output 8 or PWM input 8 would allow for an easy installation of the resistors and are rarely used in the first place.

hydra, 2016-01-27:
See #884.

SwiftingSpeed, 2016-03-17:
Thought I would leave some info if someone wants to modify the code for different ADC pins, did successfully change the PPM input pin on my Motolab Cyclone to current sensor pin.

From what I can see in the code, all ADC inputs (VBAT, RSSI, current meter and external1) needs to be on the same ADC instance on the STM processor so one have to check which ADC instance the target uses and then see which PPM/PWM inputs uses that particular ADC instance and which corresponding ADC channels they are using and the port name isn't the same as the ADC channel name so one has to reed the STM32 spec sheet to figure that out.

On the Motolab target one can see in the target.h file that instance ADC2 is used and that VBAT uses channel 2 (located on PA5) and RSSI uses channel 12 (located on PB2). There is also four lines that are out-commented which setup VBAT pin as current sensor but for Motolab this isn't usable as the VBAT pin is tied to the board battery connector powering the board.

Too look for possible pins to use, start by looking in the system/timer.c file to see what pin are mapped to inputs, if you are using a PPM receiver on a PWM capable board you should have quite a fief possible candidates and if you're using serial receiver also PPM input is free (as on Motolab board). In the code you can see that PPM input on Motolab is mapped to PA7 but we need to know what ADC instance and channel it uses and for that we need the spec sheet.

In the [spec sheet](http://www.st.com/st-web-ui/static/active/en/resource/technical/document/datasheet/DM00058181.pdf) of the STM32F303 processor there is a paragraph called "Pinouts and pin description" which first show a visual pinouts of the different packages but then also has big table listing all pin functions and by looking at the "Additional functions" column you need to find which pins that uses your target ADC instance, for pin PA7, ADC2_IN4 is listed which means ADC2 instance, channel 4 so we're in luck as it has the correct ADC instance.

To modify the target file you need to specify port (GPIOA for port A), pin (GPIO_Pin_7) and also ADC channel (ADC_Channel_4). After correctly specifying both port, pin, and channel, the PWM mapping code skips setting up the PWM input as PWM and instead uses it as ADC, changing only pin without matching the correct ADC channel won't work and maybe that's why it didn't work for you.

SwiftingSpeed, 2016-03-17:
So for the SPRACINGF3 target we have 8 PWM inputs, I have added the ADC instance and channel (if any) in parenthesis.
RC_CH2 - PA1 (ADC1_IN2)
RC_CH3 - PB11 (not an ADC input, also UART3)
RC_CH4 - PB10 (not an ADC input, also UART3)
RC_CH5 - PB4 (not an ADC input)
RC_CH6 - PB5 (not an ADC input)
RC_CH7 - PB0 (ADC3_IN12)
RC_CH8 - PB1 (ADC3_IN1)

As you can see, none of the 3 possible ADC inputs uses ADC2 instance as specified in the SPRACINGF3 target file so they are all unusable. Maybe you can instead switch ADC instance to ADC3 but then I don't know if the same ADC_DMA_CHANNEL and ADC_AHB_PERIPHERAL can be used, there's where my knowledge ends.

Looking at the output pins though we got this:
PWM1 - PA6 (ADC2_IN3)
PWM2 - PA7 (ADC2_IN4)
PWM3 - PA11 (not an ADC input)
PWM4 - PA12 (not an ADC input)
PWM5 - PB8 (not an ADC input)
PWM6 - PB9 (not an ADC input)
PWM7 - PA2 (ADC1_IN3)
PWM8 - PA3 (ADC1_IN4)

So maybe PWM 1 and 2 can be used if you enable a custom motor mapping, I don't know if something else needs to be done as the pins normally are outputs but I don't think so as the PWM setup skip pins colliding with ADC definitions and the ADC setup seems to setup the ports as inputs.

closed 2017-02-25 by hydra